PX4/PX4-SITL_gazebo-classic #49: Gimbal controller buggy

jgoppert, 2016-09-25:
The gimbal controller has some odd behaviours at different compass headings. I added an angle wrap for the error calculation, and that helped some. Also the euler angles are assumed to be 321, but I'm creating a model of the tarot t4-3d, which has angles 312, so might need to think about how best to handle that.

jgoppert, 2016-09-25:
Looks like that the wrap issue wa fixed, support for 312 would still be nice.

hsu, 2016-09-25:
For 312 support, is [QtoZXY](https://github.com/PX4/sitl_gazebo/blob/master/src/gazebo_gimbal_controller_plugin.cpp#L312) what you need?

jgoppert, 2016-09-25:
Great, seems you guys already fixed all of this and I just didn't pull recently on my branch.

closed 2016-09-25 by jgoppert