PX4/PX4-SITL_gazebo-classic #56: PX4 simulation in Gazebo

DeanJJJ, 2016-10-17:
anyone know how to simulate PX4 in Gazebo? when I enter "make posix_sitl_default gazebo", the error which is 
CMakeFiles/Makefile2:9960: recipe for target 'src/firmware/posix/CMakeFiles/gazebo.dir/rule' failed
make[2]: *** [src/firmware/posix/CMakeFiles/gazebo.dir/rule] Error 2
Makefile:3497: recipe for target 'gazebo' failed
make[1]: *** [gazebo] Error 2
Makefile:160: recipe for target 'posix_sitl_default' failed
make: *** [posix_sitl_default] Error 2
will appear.

julianoes, 2016-10-17:
@DeanJJJ Can you please post the whole output of make and the output of

```
git rev-parse HEAD
```

DeanJJJ, 2016-10-17:
@julianoes thank you for reply! The following is the output of ' git rev-parse HEAD '
db44129ec099a05debf9187da2fd09035c9a67d7

DeanJJJ, 2016-10-17:
@julianoes and the following is the while output of make 
PX4 CONFIG: /home/ubuntu/src/Firmware/build_posix_sitl_default
[  0%] Built target git_genmsg
[  0%] Built target prebuild_targets
[  1%] Built target git_gencpp
[  1%] Performing configure step for 'sitl_gazebo'
[  1%] Built target git_uavcan
[  2%] Built target git_mavlink
[  2%] Built target git_matrix
-- install-prefix: /usr
[  2%] Built target git_driverframework
[  2%] Built target git_jmavsim
[  2%] Built target git_ecl
[  4%] Built target df_driver_framework
[  4%] Built target git_gazebo
[  4%] Built target mixer_gen
[  5%] Built target xml_gen
[ 24%] Built target msg_gen
[ 25%] Built target platforms__common
[ 26%] Built target generate_topic_listener
CMake Error at CMakeLists.txt:23 (find_package):
  By not providing "Findgazebo.cmake" in CMAKE_MODULE_PATH this project has
  asked CMake to find a package configuration file provided by "gazebo", but
  CMake did not find one.

  Could not find a package configuration file provided by "gazebo" with any
  of the following names:

    gazeboConfig.cmake
    gazebo-config.cmake

  Add the installation prefix of "gazebo" to CMAKE_PREFIX_PATH or set
  "gazebo_DIR" to a directory containing one of the above files.  If "gazebo"
  provides a separate development package or SDK, be sure it has been
  installed.

-- Configuring incomplete, errors occurred!
See also "/home/ubuntu/src/Firmware/build_posix_sitl_default/build_gazebo/CMakeFiles/CMakeOutput.log".
src/firmware/posix/CMakeFiles/sitl_gazebo.dir/build.make:105: recipe for target 'external/Stamp/sitl_gazebo/sitl_gazebo-configure' failed
make[4]: *** [external/Stamp/sitl_gazebo/sitl_gazebo-configure] Error 1
CMakeFiles/Makefile2:9086: recipe for target 'src/firmware/posix/CMakeFiles/sitl_gazebo.dir/all' failed
make[3]: *** [src/firmware/posix/CMakeFiles/sitl_gazebo.dir/all] Error 2
make[3]: *** Waiting for unfinished jobs....
[ 26%] Built target drivers__boards__sitl
[ 28%] Built target drivers__device
[ 35%] Built target systemcmds__tests
CMakeFiles/Makefile2:9960: recipe for target 'src/firmware/posix/CMakeFiles/gazebo.dir/rule' failed
make[2]: *** [src/firmware/posix/CMakeFiles/gazebo.dir/rule] Error 2
Makefile:3497: recipe for target 'gazebo' failed
make[1]: *** [gazebo] Error 2
Makefile:160: recipe for target 'posix_sitl_default' failed
make: *** [posix_sitl_default] Error 2

LorenzMeier, 2016-10-17:
You don't have all Gazebo packages installed. Please go back to the PX4 Linux installation instructions and make sure to install *everything*.

julianoes, 2016-10-17:
@DeanJJJ please update to latest master. There have been fixes for gazebo:

https://github.com/PX4/Firmware/commit/2d039af7a4299c6565e0d47f333adad65fd0c0a0

DeanJJJ, 2016-10-17:
@julianoes  thx a lot, I will try it.

DeanJJJ, 2016-10-21:
@julianoes sorry for bothering again. I have updated firmware to latest master and installed Gazebo packages step by step. However, new errors appear after 100% building. Errors is shown below:

[100%] Completed 'sitl_gazebo'
[100%] Built target sitl_gazebo
Scanning dependencies of target gazebo
args: /home/ubuntu/src/Firmware/build_posix_sitl_default/src/firmware/posix/px4 posix-configs/SITL/init/lpe none gazebo none /home/ubuntu/src/Firmware /home/ubuntu/src/Firmware/build_posix_sitl_default
SITL ARGS
sitl_bin: /home/ubuntu/src/Firmware/build_posix_sitl_default/src/firmware/posix/px4
rcS_dir: posix-configs/SITL/init/lpe
debugger: none
program: gazebo
model: none
src_path: /home/ubuntu/src/Firmware
build_path: /home/ubuntu/src/Firmware/build_posix_sitl_default
empty model, setting iris as default
/home/ubuntu/src/Firmware/Tools/sitl_run.sh: line 58: kill: ubuntu: arguments must be process or job IDs
src/firmware/posix/CMakeFiles/gazebo.dir/build.make:57: recipe for target 'src/firmware/posix/CMakeFiles/gazebo' failed
make[4]: *** [src/firmware/posix/CMakeFiles/gazebo] Error 1
CMakeFiles/Makefile2:10004: recipe for target 'src/firmware/posix/CMakeFiles/gazebo.dir/all' failed
make[3]: *** [src/firmware/posix/CMakeFiles/gazebo.dir/all] Error 2
CMakeFiles/Makefile2:10011: recipe for target 'src/firmware/posix/CMakeFiles/gazebo.dir/rule' failed
make[2]: *** [src/firmware/posix/CMakeFiles/gazebo.dir/rule] Error 2
Makefile:3510: recipe for target 'gazebo' failed
make[1]: *** [gazebo] Error 2
Makefile:160: recipe for target 'posix_sitl_default' failed
make: *** [posix_sitl_default] Error 2

julianoes, 2016-10-21:
Debug what goes wrong here:
```
/home/ubuntu/src/Firmware/Tools/sitl_run.sh: line 58: kill: ubuntu: arguments must be process or job IDs
```

DeanJJJ, 2016-10-24:
@julianoes it works, thank you

closed 2016-10-24 by julianoes

xaviruvpadhiyar98, 2017-02-18:
CMakeFiles/Makefile2:6216: recipe for target 'modules/videoio/CMakeFiles/opencv_videoio.dir/all' failed
make[1]: *** [modules/videoio/CMakeFiles/opencv_videoio.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....

xaviruvpadhiyar98, 2017-02-18:
make -j4
[  1%] Built target opencv_core_pch_dephelp
[  1%] Built target opencv_perf_core_pch_dephelp
[  3%] Built target libwebp
[  4%] Built target opencv_imgcodecs_pch_dephelp
[  4%] Built target opencv_imgproc_pch_dephelp
[  4%] Built target opencv_videoio_pch_dephelp
[  5%] Built target opencv_highgui_pch_dephelp
[  6%] Built target IlmImf
[  6%] Built target opencv_ts_pch_dephelp
[  6%] Built target opencv_test_core_pch_dephelp
[  6%] Built target opencv_test_flann_pch_dephelp
[  8%] Built target opencv_flann_pch_dephelp
[  8%] Built target opencv_perf_imgproc_pch_dephelp
[  8%] Built target opencv_test_imgproc_pch_dephelp
[  9%] Built target opencv_test_ml_pch_dephelp
[  9%] Built target opencv_ml_pch_dephelp
[  9%] Built target opencv_photo_pch_dephelp
[  9%] Built target opencv_perf_photo_pch_dephelp
[  9%] Built target opencv_reg_pch_dephelp
[  9%] Built target opencv_test_photo_pch_dephelp
[  9%] Built target opencv_test_reg_pch_dephelp
[  9%] Built target opencv_perf_reg_pch_dephelp
[  9%] Built target opencv_surface_matching_pch_dephelp
[  9%] Built target opencv_video_pch_dephelp
[  9%] Built target opencv_test_video_pch_dephelp
[  9%] Built target opencv_perf_video_pch_dephelp
[  9%] Built target opencv_dnn_pch_dephelp
[  9%] Built target opencv_test_dnn_pch_dephelp
[  9%] Built target opencv_fuzzy_pch_dephelp
[ 10%] Built target libprotobuf
[ 10%] Built target opencv_test_fuzzy_pch_dephelp
[ 11%] Built target opencv_perf_imgcodecs_pch_dephelp
[ 11%] Built target opencv_test_imgcodecs_pch_dephelp
[ 12%] Built target opencv_shape_pch_dephelp
[ 12%] Built target opencv_test_shape_pch_dephelp
[ 12%] Built target opencv_test_videoio_pch_dephelp
[ 12%] Built target opencv_perf_videoio_pch_dephelp
[ 12%] Built target opencv_test_highgui_pch_dephelp
[ 12%] Built target opencv_objdetect_pch_dephelp
[ 12%] Built target opencv_perf_objdetect_pch_dephelp
[ 12%] Built target opencv_test_objdetect_pch_dephelp
[ 12%] Built target opencv_superres_pch_dephelp
[ 12%] Built target opencv_plot_pch_dephelp
[ 12%] Built target opencv_perf_superres_pch_dephelp
[ 12%] Built target opencv_test_superres_pch_dephelp
[ 12%] Built target opencv_xobjdetect_pch_dephelp
[ 13%] Built target opencv_test_xphoto_pch_dephelp
[ 13%] Built target opencv_perf_xphoto_pch_dephelp
[ 13%] Built target opencv_bgsegm_pch_dephelp
[ 13%] Built target opencv_test_bgsegm_pch_dephelp
[ 13%] Built target opencv_bioinspired_pch_dephelp
[ 13%] Built target opencv_test_bioinspired_pch_dephelp
[ 13%] Built target opencv_perf_bioinspired_pch_dephelp
[ 13%] Built target opencv_dpm_pch_dephelp
[ 13%] Built target opencv_face_pch_dephelp
[ 13%] Built target opencv_features2d_pch_dephelp
[ 13%] Built target opencv_test_features2d_pch_dephelp
[ 13%] Built target opencv_perf_features2d_pch_dephelp
[ 13%] Built target opencv_perf_line_descriptor_pch_dephelp
[ 13%] Built target opencv_line_descriptor_pch_dephelp
[ 13%] Built target opencv_test_line_descriptor_pch_dephelp
[ 13%] Built target opencv_saliency_pch_dephelp
[ 13%] Built target opencv_text_pch_dephelp
[ 13%] Built target opencv_calib3d_pch_dephelp
[ 13%] Built target opencv_test_calib3d_pch_dephelp
[ 15%] Built target opencv_perf_calib3d_pch_dephelp
[ 16%] Built target opencv_ccalib_pch_dephelp
[ 16%] Built target opencv_rgbd_pch_dephelp
[ 16%] Built target opencv_test_rgbd_pch_dephelp
[... 105 lines cut ...]
[ 26%] Built target pch_Generate_opencv_test_optflow
[ 26%] Built target pch_Generate_opencv_optflow
[ 26%] Built target pch_Generate_opencv_test_stitching
[ 26%] Built target pch_Generate_opencv_stitching
[ 26%] Built target pch_Generate_opencv_perf_stitching
[ 29%] Built target opencv_core
[ 29%] Built target opencv_flann
[ 30%] Built target opencv_ml
[ 30%] Built target opencv_surface_matching
[ 32%] Built target opencv_imgproc
[ 33%] Built target opencv_reg
[ 34%] Built target opencv_video
[ 36%] Built target opencv_imgcodecs
[ 37%] Built target opencv_photo
make[2]: *** No rule to make target '/usr/include/../lib/libavcodec.a', needed by 'lib/libopencv_videoio.so.3.1.0'.  Stop.
CMakeFiles/Makefile2:6216: recipe for target 'modules/videoio/CMakeFiles/opencv_videoio.dir/all' failed
make[1]: *** [modules/videoio/CMakeFiles/opencv_videoio.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
[ 37%] Built target opencv_fuzzy
[ 37%] Built target opencv_shape
[ 38%] Built target opencv_dnn
Makefile:160: recipe for target 'all' failed
make: *** [all] Error 2

I m stuck here plz help me

kevinuav, 2020-03-14:
Hey,guys!
    I encounter an error similar to this issue when I compile for the gazebo with branch master in my  Mac OS X(10.15.3 Catalina).What's the problem?Thank you!

here is the message from terminal:
%make px4_sitl gazebo_standard_vtol
.......
[14/98] Generating /Users/kevin/Firmware/Tools/sitl_gazebo/models/iris/iris.sdf
FAILED: /Users/kevin/Firmware/Tools/sitl_gazebo/models/iris/iris.sdf 
cd /Users/kevin/Firmware/Tools/sitl_gazebo && rm -f /Users/kevin/Firmware/Tools/sitl_gazebo/models/iris/iris.sdf && /usr/local/bin/python3 /Users/kevin/Firmware/Tools/sitl_gazebo/scripts/xacro.py -o 
.......
[705/716] Linking CXX static library src/lib/flight_tasks/libFlightTasks.a
FAILED: external/Stamp/sitl_gazebo/sitl_gazebo-build 
cd /Users/kevin/Firmware/build/px4_sitl_default/build_gazebo && /usr/local/Cellar/cmake/3.16.5/bin/cmake --build .
......
ninja: build stopped: subcommand failed.
make: *** [px4_sitl] Error 1

Francis-Herman, 2020-07-02:
home/francis/gpstk/dev/ext/lib/Math/Vector/VectorOperators.hpp:124:7: note: in expansion of macro ‘VecBaseNewBinaryOperator’
  124 |       VecBaseNewBinaryOperator(!=, Vector<bool>)
      |       ^~~~~~~~~~~~~~~~~~~~~~~~
/home/francis/gpstk/dev/ext/lib/Math/Vector/VectorOperators.hpp:104:8: note:   template argument deduction/substitution failed:
  104 | retval operator func(const T l, const ConstVectorBase<T, BaseClass>& r) \
      |        ^~~~~~~~
/home/francis/gpstk/dev/ext/lib/Math/Vector/VectorOperators.hpp:124:7: note: in expansion of macro ‘VecBaseNewBinaryOperator’
hello! 
how do i overcome this error please 

 124 |       VecBaseNewBinaryOperator(!=, Vector<bool>)
      |       ^~~~~~~~~~~~~~~~~~~~~~~~
/home/francis/gpstk/dev/ext/apps/geomatics/JPLeph/testSSEph.cpp:243:18: note:   ‘std::ostream’ {aka ‘std::basic_ostream<char>’} is not derived from ‘const gpstk::ConstVectorBase<T, BaseClass>’
  243 |    if(LOGstrm != cout) cout << PrgmName << " timing: " << fixed << setprecision(9)
      |                  ^~~~
make[2]: *** [ext/apps/geomatics/JPLeph/CMakeFiles/testSSEph.dir/build.make:63: ext/apps/geomatics/JPLeph/CMakeFiles/testSSEph.dir/testSSEph.cpp.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:1058: ext/apps/geomatics/JPLeph/CMakeFiles/testSSEph.dir/all] Error 2
make: *** [Makefile:152: all] Error 2

Francis-Herman, 2020-07-02:
francis@francis-HP-Laptop-15-bs0xx:~/Downloads/source$ ./correctScript
bash: ./correctScript: No such file or directory

how do install the software inside the source folder

julianoes, 2020-07-06:
Please create new issues and provide all the necessary context, and don't necrobump old threads, thanks.