PX4/PX4-SITL_gazebo-classic #59: how to insert another PX4 model on gazebo?

cynthiamoon, 2016-11-15:
I want to insert another PX4 model on gazebo. mavros also can recognize it.Anyone know how can i do?

mickey13, 2016-11-26:
Is there an example of how to get this running properly?  In my case I have a mavros_node and px4 executable running for each model.  What I'm seeing is the Gazebo models flying around erratically when one of them is given a takeoff command.  I'm guessing this is due to both Gazebo models using the same PX4 node.  I've been trying to set up the UDP ports for each model / mavros / PX4 appropriately, but I'm not having much luck.  Any ideas or working examples?  Thanks.

imcnanie, 2016-12-16:
Wrote a swarm simulator that runs pong a while ago. We had a basic simulator that worked with Rviz. tfoote made a PR to get multiple PX4 gazebo simulators in it: https://github.com/3drobotics/swarm/pull/1

julianoes, 2016-12-16:
There is this file: https://github.com/PX4/Firmware/blob/master/Tools/sitl_multiple_run.sh

Not sure where the docs are, or if and how it works.

lwcworld, 2016-12-27:
Can anyone give an example step by step?

cynthiamoon, 2017-01-06:
@mickey13 how to add another autopilots on one SITL gazebo.

mickey13, 2017-01-07:
@cynthiamoon You might need to make a copy of px4_firmware/posix-configs/SITL/init/<model> for each of your models and set the UDP ports manually for each model that you have in the simulation.  Not sure what particular issue you're seeing at the moment, but this was a hurdle for me.

cynthiamoon, 2017-01-07:
@mickey13 Can you implement multi - aircraft control on gazebo？

tfoote, 2017-01-18:
@cynthiamoon see https://github.com/osrf/uctf where we're simulating many vehicles with one gazebo instance. Each drone gets it's own plugin with specific ports, and the ardupilot connects to the plugin via udp.

cynthiamoon, 2017-01-24:
@tfoote  can I run this project use ROS indigo and Gazebo6 ？

tfoote, 2017-01-24:
Early versions were tested on gazebo7. There are some changes in gazebo8 that make it much easier. In it's current state I wouldn't suggest trying to use it on older versions of gazebo. But we plan to turn it into a pure plugin which might be backportable in the future.

cynthiamoon, 2017-02-08:
@tfoote I read the tutorial.  https://github.com/osrf/uctf/blob/master/doc/install_binary/readme.md
terminial display“unable to locate package sasc-gazebo-sitl”,after i run "sudo apt install sasc-gazebo-sitl"
What should I do next？

tfoote, 2017-02-08:
@cynthiamoon That means you either have not added the package to your path or you're not on xenial. For the binary installs we only support xenial.

leemaker48, 2017-04-05:
Does anyone have success??
I succeeded showing two-iris on the one-world screen. but,  if i give the arming command on terminal,  Two-iris are arming at the same time. What i means that udp port is separated but, these are tied up to receive commands at the same time. How can i add udp-path??? Recently, i checked the master-version on /sitl_gazebo/src/gazebo_mavlink_interface.cpp file, UDP allocation method has changed by element(shared-lib). 'mavlink_udp_port_' is element in '_sdf' object, isn't it?
How can i create a element? something likes 'mavlink_udp_port_1,,2..blahblah..'
i think ,,element parts is very important things for multiple_sitl on gazebo.

thanks.
leemaker48.

mzahana, 2017-05-13:
Here is a PR, in PX4 repo, that solves this issue,
https://github.com/PX4/Firmware/pull/7235

TSC21, 2018-01-04:
Solved in https://github.com/PX4/Firmware/pull/7235. Check devguide docs also on how to proceed.

closed 2018-01-04 by TSC21

YCH188, 2018-03-03:
@tfoote I run as the readme.md in uctf. But when I type "/opt/sasc/bin/fti_env.sh fti.py -d enp0s25 -z". The terminal shows "WARNING: With great power comes great responsibility!
Couldn't start up socket on interface 'enp0s25'"
And I found no similar sloution on google for this. Wishing for your help!

tfoote, 2018-03-07:
@YCH188 if this is specific to the uctf repo you should probably open a ticket there. 'enp0s25' is the default ethernet device on our testing machine. You'll need to modify that to reference your network device.