PX4/PX4-SITL_gazebo-classic #63: CI sever needs some new packages

jgoppert, 2016-12-02:
http://sitl01.dronetest.io/job/PX4%20Firmware%20POSIX%20SITL%20PRs/3050/console

@dagar

dagar, 2016-12-02:
FYI this is running within https://hub.docker.com/r/px4io/px4-dev-ros/ which is defined here - https://github.com/PX4/containers/blob/master/docker/px4-dev/DockerfileRos

I'll start with just gstreamer-base-1.0, unless you happen to know the exact dependencies. There are a lot of gstreamer packages...

dagar, 2016-12-03:
I added the packages, fixed px4io/px4-dev-ros, and may have broken all of our SITL tests in the process. 
Due to a ppa change px4io/px4-dev-ros had stopped building successfully ~6 months ago.
Standby...

jgoppert, 2016-12-03:
@dagar OK, great, was actually trying to build it locally as we speak

dagar, 2016-12-04:
FYI The last working ROS docker image was restored and tagged on Jenkins for now. I'm updating all the images (https://github.com/PX4/containers/tree/master/docker/px4-dev) to extend a common PX4 base, which will bring px4-dev-ros to 16.04 and newer ROS in the process.

LorenzMeier, 2016-12-05:
Jenkins is currently not reporting any status - is this related?

LorenzMeier, 2016-12-05:
@AndreasAntener FYI

AndreasAntener, 2016-12-05:
This should not be related

dagar, 2017-01-14:
@jgoppert what would you like to do here?
Do you want to try the latest px4-dev-ros and get it working with your testing? https://hub.docker.com/r/px4io/px4-dev-ros/

Last I heard the ros based mission tests (http://sitl01.dronetest.io/job/PX4%20Firmware%20POSIX%20SITL%20PRs/) still only worked reliably with the ROS container built ~ 7 months ago.

dagar, 2017-01-15:
px4-dev-simulation (gazebo, jmavsim) and px4-dev-ros (simulation + ROS) have gstreamer now. https://github.com/PX4/containers/blob/master/docker/px4-dev/Dockerfile_simulation#L14

Our testing still needs to be updated to use the new environments.

closed 2017-01-15 by dagar