PX4/PX4-SITL_gazebo-classic #67: Typhoon H480 Oscillation

imcnanie, 2016-12-16:
As soon as I remove the legs from the Typhoon H480 sdf model, the simulation oscillates a lot more. Here's a before and after:

Legs:
[image]

No Legs:
[image]

SDF for reference:
[hexa.sdf.txt](https://github.com/PX4/sitl_gazebo/files/656201/hexa.sdf.txt)

Any idea why? I'm using the Typhoon as the base for my hexacopter, but mine doesn't have legs or a gimbal. Thank you!

imcnanie, 2016-12-16:
Actually, increased the mass of the copter to account for the loss of weight and got a different oscillation (makes sense). Should I do an autotune? Or are there other properties to watch for?

nicolaerosia, 2017-01-10:
Just wondering if you figured this one out

imcnanie, 2017-01-11:
Yes, it looks like the SDF (in Tools/sitl_gazebo/models/typhoon_h480/typhoon_h480.sdf) defined the camera and legs as fairly heavy objects, so when I removed them, the extra weight changed the tune. You can fix this by adding the lost weight to the rest of the model (in the sdf) or changing the PIDs.

closed 2017-08-16 by ChristophTobler