PX4/PX4-SITL_gazebo-classic #78: Questions about implementing a new copter configuration for sitl_gazebo

imcnanie, 2017-01-11:
I'm planning on implementing a new copter configuration for the sitl_gazebo plugin, I have a few questions about some of the plugin variables:

How are the motorConstant and motorMoment variables calculated? (in the SDF):
https://github.com/PX4/sitl_gazebo/blob/d6c12fb9e68608db2ad56f4ec6c74e5a491e17ee/src/gazebo_motor_model.cpp#L186

Similarly curious about rotorDragCoefficient, and rollingMomentCoefficient. Are these parameters following a paper? Also, curious what unit maxRotVelocity is in.

TSC21, 2017-07-15:
@imcnanie were you able to get the answers to this?

imcnanie, 2017-10-02:
Yes! Fantastic! I mostly figured it out, but that's much more comprehensive. We can close this now.

ChristophTobler, 2017-10-03:
@imcnanie Anything that needs to be documented?

TSC21, 2018-01-04:
> @imcnanie Anything that needs to be documented?

What needs documentation is #110, which will answer to this. I'm closing this so we can follow up on the other issue. @imcnanie can you leave your finding there, please? Thanks

closed 2018-01-04 by TSC21