PX4/PX4-SITL_gazebo-classic #80: . /usr/share/mavlink_sitl_gazebo/setup.sh No such file or directory

saimouli, 2017-01-15:
I did check my usr/share folder and found only two folders related to gazebo titled as "Gazebo"->setup.sh  and "Gazebo7"-> setup.sh, worlds,models etc. I couldn't find any other. 

Also when I execute gazebo worlds/iris.world, empty world pops up.

Maunsi, 2017-08-22:
Did you figure out how to solve it? I also don't have a mavlink_sitl_gazebo folder

TSC21, 2017-08-22:
mavlink_stil_gazebo is generated when you build your workspace. So you put your PX4 Firmware folder onto `~/ws/src` and build the code. `mavlink_sitl_gazebo` will be generated that way. `setup.sh` will be on `/ws/build/mavlink_sitl_gazebo`. This can be generated through CMake build or through `catkin tools` (in conjunction with ROS).

Maunsi, 2017-08-22:
Thank you!!
I was using roslaunch to launch a file and I was not building prior to that, hence the problem.

closed 2017-08-22 by TSC21