PX4/PX4-SITL_gazebo-classic #86: plane motor issue

dagar, 2017-02-07:
Something strange happens to the plane model in auto landing at flare. The throttle cuts (in the px4 logs), but the vehicle starts climbing and gets stuck in a flyaway.

[image]

The only thing noteworthy I saw in the console is this message.

    [Wrn] [Publisher.cc:141] Queue limit reached for topic /gazebo/default/plane/gazebo/command/motor_speed, deleting message. This warning is printed only once.

http://logs.uaventure.com/view/xaJwxSYbc7jXBN2Xa7Yuii

dagar, 2017-02-08:
It looks like this is probably coming from the pwm_out_sim in PX4.

LorenzMeier, 2017-02-13:
Fixed.

closed 2017-02-13 by LorenzMeier