PX4/PX4-SITL_gazebo-classic #93: Failed to call the plugin function from a xacro file.

yajingwang94, 2017-03-12:
Hi, 
I used to use this package with a rotorcraft model of `SDF`, which worked well. But when I tried to add a fixed_wing plane model of `xacro` formation to the `sitl_gazebo` package, it kept reporting errors.

A specialist one is as follows(which was reported after I inserted the `mavlink_interface_macro` to my xacro model description file):

`when processing file: ``/home/u23/WorkSpace/03.fixedwing_sim_pix/04.Firmware/Firmware/Tools/sitl_gazebo/models/techpod/techpod_base.xacro`
`unknown macro name: xacro:mavlink_interface_macro`
`when processing file: ``/home/u23/WorkSpace/03.fixedwing_sim_pix/04.Firmware/Firmware/Tools/sitl_gazebo/models/techpod/techpod_base.xacro`
`while processing ``/home/u23/WorkSpace/03.fixedwing_sim_pix/04.Firmware/Firmware/launch/spawn_fixed_wing.launch:`
`Invalid <param> tag: Cannot load command parameter [robot_description]: command [     /opt/ros/jade/share/xacro/xacro.py ``'/home/u23/WorkSpace/03.fixedwing_sim_pix/04.Firmware/Firmware/Tools/sitl_gazebo/models/techpod/techpod_base.xacro'     enable_logging:=false     enable_ground_truth:=true     log_file:=techpod     wait_to_record_bag:=false     uav_name:=techpod     namespace:=techpod] returned with code [2]. `

`Param xml is <param command="     $(find xacro)/xacro.py '$(arg model)'     enable_logging:=$(arg enable_logging)     enable_ground_truth:=$(arg enable_ground_truth)     log_file:=$(arg log_file)     wait_to_record_bag:=$(arg wait_to_record_bag)     uav_name:=$(arg uav_name)     namespace:=$(arg namespace)" name="robot_description"/>`
`The traceback for the exception was written to the log file`

I checked the whole process but had no idea where I did wrong.
 I wonder, is it acceptable that I provide a` xacro `source file for the `gazebo_mavlink_interface.cpp` as the value of `sdf::ElementPtr _sdf`?

mzahana, 2017-05-13:
Why don't you try to convert xacro to sdf as follows,
```
rosrun xacro xacro.py your_xacro.xacro > your_urdf.urdf
gz sdf -p your_urdf.urdf > your_sdf.sdf
```
Then use the sdf file.

TSC21, 2018-01-04:
Assuming this is solved.

closed 2018-01-04 by TSC21