PX4/PX4-SITL_gazebo-classic #102: Gazebo topics not getting published when running any SITL simulation

ganeshrk95, 2017-05-25:
I have the latest master clones. When I run the basic IRIS SITL using:

`make posix_sitl_default gazebo`

After the model loads and PX4 shell opens, I did:

`gz topic -l `

I get the following
[image]

I just cannot echo any iris related topics using:

`gz topic -e /gazebo/default/iris/$NAME `

It returns nothing. I'm developing a Tiltrotor Tricopter model for SITL. I need to view what's going into the motor_speed topics to debug my code. 

I keep getting these warnings but I think this shouldn't cause the topics to not publish anything.
[image]

wil3, 2017-08-11:
I'm actually looking into this right. I was attempting to use pygazebo to control the simulator however when you publish any messages it crashes gzserver. 

`gzserver: /home/wil/workspace/PX4/Firmware/Tools/sitl_gazebo/src/gazebo_imu_plugin.cpp:190: void gazebo::GazeboImuPlugin::addNoise(Eigen::Vector3d*, Eigen::Vector3d*, double): Assertion `dt > 0.0' failed.`

Looking into this closer (at least for my case) the simulation time changes and the assertion fails. Working on a fix now.

TSC21, 2018-01-04:
@ganeshrk95 is this solved to you?

ganeshrk95, 2018-03-21:
Still have this issue and can't figure out what is causing this. Increasing the queue limit doesn't seem to fix this.

TSC21, 2018-05-31:
The problem is this: https://bitbucket.org/osrf/gazebo/issues/1149/echo-topic-with-custom-messages-doesnt. And it seems it is still not fixed. So this is not related with `sitl_gazebo` but rather with Gazebo itself. @ganeshrk95 have you tried a further version, like Gazebo 8 or 9 and check if this is still happens?

TSC21, 2018-05-31:
On the Bitbucket issue it is described to hotfixes for it. But I don't actually think they are worth it if the only thing we want from this is to echo the topics.

ganeshrk95, 2018-06-28:
It did happen with Gazebo 8 if I remember correctly.

TSC21, 2018-10-05:
This is Gazebo internals related, not related to any plugin on the this repository. Closing

closed 2018-10-05 by TSC21