PX4/PX4-SITL_gazebo-classic #103: Rotors stop moving when attachend to another moving frame

SwapZap, 2017-06-11:
Hi,
I am simulating a VTOL tiltrotor. I used the `standard_vtol` as a template to get started. I have attached a link called `shaft` to the `base_link`. The `shaft` link is getting control from channel 4 for tilt servo. Front rotors are attached to the `shaft`. Now when I use this setup the rotors stop rotating. Notably, when the control on `shaft` is removed the rotors start spinning.
Any suggestions on how to get both of them working?

TSC21, 2018-01-04:
@SwapUNaph any updates on this?

SwapUNaph, 2018-03-04:
I have done a hack.
Now I am passing the tilt angle to the model and calculating the component of force vector in the forward(X) as well as upward(Z) direction.

RomanBapst, 2018-03-04:
@SwapUNaph I have pushed a tiltrotor model to PX4 with the same idea as how you described it above.
It took some time to get it working and I think I also had the same issue at some point. If I remember correctly it's not advisable to directly try to set the angle of a control joint but rather apply a moment to it and use a PID controller to control the angle.

TSC21, 2018-03-04:
I think we can close this now as this as a solution was already found by @RomanBapst.

closed 2018-03-04 by TSC21