PX4/PX4-SITL_gazebo-classic #104: Gazebo Wind Plugin Protobuf Error

eric1221bday, 2017-06-13:
Hi

I've recently decided to try and compile the site_gazebo package on Mac. This entailed building it against Gazebo 8 as well as Protobuf 3.3.0 and protobuf-c 1.2.1. Aside from the expected Gazebo 8 compatibility issues being addressed in the pull request #95, I've also come across a protobuf error in regards to the wind photo file.

```shell
/Users/eric1221bday/sandbox/quadcopter/src/sitl_gazebo/src/gazebo_wind_plugin.cpp:25:10: warning: non-portable path to file '"wind.pb.h"'; specified path differs in case from file name on disk [-Wnonportable-include-path]
#include "Wind.pb.h"
         ^~~~~~~~~~~
         "wind.pb.h"
/Users/eric1221bday/sandbox/quadcopter/src/sitl_gazebo/src/gazebo_wind_plugin.cpp:82:39: error: use of undeclared identifier 'wind_msgs'
  wind_pub_ = node_handle_->Advertise<wind_msgs::msgs::Wind>(wind_pub_topic_, 1);
                                      ^
/Users/eric1221bday/sandbox/quadcopter/src/sitl_gazebo/src/gazebo_wind_plugin.cpp:82:61: error: expected unqualified-id
  wind_pub_ = node_handle_->Advertise<wind_msgs::msgs::Wind>(wind_pub_topic_, 1);
                                                            ^
/Users/eric1221bday/sandbox/quadcopter/src/sitl_gazebo/src/gazebo_wind_plugin.cpp:105:3: error: use of undeclared identifier 'wind_msgs'
  wind_msgs::msgs::Wind wind_msg;
```
Some digging has revealed that because gazebo 8 added a simple Wind class that is different from the currently generated wind class within site_gazebo, there has been a namespace collision resulting in the current `#include "Wind.pb.h"` mistakening the site_gazebo wind header for the gazebo 8 one.

I've opened this issue for preliminary discussion as to how we might resolve this namespace collision.

eric1221bday, 2017-06-13:
So one solution that I've been able to implement was to rename the `Wind.proto` file to `RotorWind.proto`. I kept the message name and namespace the same. This way when including the headers catkin and make would no longer be confused. However I am unsure of the suitability of naming a photo filename and contained message differently.

TSC21, 2018-01-04:
@eric1221bday does this still subsist?

TSC21, 2018-05-30:
Markings as solved, though it would be important to reboot the wind plugin.

closed 2018-05-30 by TSC21