PX4/PX4-SITL_gazebo-classic #111: GStreamer Camera Plugin failed to enable

whyscience, 2017-07-16:
First enable in CMakelists
`option(BUILD_GSTREAMER_PLUGIN "enable gstreamer plugin" "OFF")`
Then edit `typhoon_h480.sdf`:
```xml
        <plugin name="cgo3_camera_controller" filename="libgazebo_ros_camera.so">
          <alwaysOn>true</alwaysOn>
          <updateRate>0.0</updateRate>
          <cameraName>cgo3_camera</cameraName>
          <imageTopicName>image_raw</imageTopicName>
          <cameraInfoTopicName>camera_info</cameraInfoTopicName>
          <frameName>cgo3_camera_optical_frame</frameName>
          <hackBaseline>0.0</hackBaseline>
          <distortionK1>0.0</distortionK1>
          <distortionK2>0.0</distortionK2>
          <distortionK3>0.0</distortionK3>
          <distortionT1>0.0</distortionT1>
          <distortionT2>0.0</distortionT2>
        </plugin>
        -->
        <!-- GStreamer camera plugin (needs a lot of CPU! Consider lowering the
             camera image size when enabling) -->
        <plugin name="GstCameraPlugin" filename="libgazebo_gst_camera_plugin.so">
            <robotNamespace></robotNamespace>
            <udpPort>5600</udpPort>
        </plugin>
        <plugin name="GeotaggedImagesPlugin" filename="libgazebo_geotagged_images_plugin.so">
            <robotNamespace></robotNamespace>
            <interval>1</interval>
            <width>3840</width>
            <height>2160</height>
        </plugin>
```

but got this problem
`[Err] [Plugin.hh:165] Failed to load plugin libgazebo_ros_camera.so: libCameraPlugin.so: cannot open shared object file: No such file or directory`
[image]

libgazebo_ros_camera.so is in `/opt/ros/kinetic/lib`
libCameraPlugin.so is in `/usr/lib/x86_64-linux-gnu/gazebo-7/plugins`

add PATH in sh
`# setup Gazebo env and update package path
export GAZEBO_PLUGIN_PATH=/opt/ros/kinetic/lib:${GAZEBO_PLUGIN_PATH}
export GAZEBO_PLUGIN_PATH=/usr/lib/x86_64-linux-gnu/gazebo-7/plugins:${GAZEBO_PLUGIN_PATH}
export GAZEBO_PLUGIN_PATH=${BUILD_DIR}/build_gazebo:${GAZEBO_PLUGIN_PATH}`
set this does not help

What's the problem?

TSC21, 2018-01-04:
@whyscience please check if this is solved to you using upstream master. If not, we reopen the issue. Closing now.

closed 2018-01-04 by TSC21

YCH188, 2018-02-04:
I also have this problems, and haven't found a solution yet.

YCH188, 2018-02-04:
@whyscience @TSC21   Oh, my problem is failed to load plugin libgazebo_gst_camera_plugin.so