PX4/PX4-SITL_gazebo-classic #112: ROS bulid failure

jgoppert, 2017-07-17:
```bash
src/mavlink_sitl_gazebo/include/gazebo_geotagged_images_plugin.h:32:41: fatal error: mavlink/v2.0/common/mavlink.h: No such file or directory
```

Some include path issue, tracking it down now, wish we had CI on this.

closed 2017-07-17 by jgoppert

eric1221bday, 2017-07-29:
I've also encountered this issue, however your fix didn't work, primarily I suspect because I am currently build mavros and mavlink from scratch. As such roscpp include dirs will not contain my mavros and mavlink installations. If Mavlink is to be removed from the repo I think a fix involving catkin would be far more versatile than just using cmake.

canberkgurel, 2020-11-10:
The missing header file has been moved from `common` to `minimal`.

commit d1da926f (2017-07-17): Add roscpp include path to build to find mavros, closes #112