PX4/PX4-SITL_gazebo-classic #113: how to starting Gazebo and PX4 separately

TokyoClod, 2017-07-26:
when i echo the code `make posix_sitl_default gazebo`,the virtual world and the default PX4 show together
i'm wondering how to starting Gazebo and PX4 separately,and modify the virtual world.
the code `make posix_sitl_default gazebo_none_ide`in PX4 Developer Guide doesn't work

EliaTarasov, 2018-02-06:
You can run gazebo independently (but with all plugins that are needed to start copter model)
like this:
cd workspace/Firmware
export GAZEBO_PLUGIN_PATH=${GAZEBO_PLUGIN_PATH}:build/posix_sitl_default/build_gazebo
gazebo Tools/sitl_gazebo/worlds/iris.world --verbose

Then in separate window run: make posix_sitl_default gazebo no_sim=1

Now you have px4 and gazebo running separately

hamishwillee, 2018-02-07:
@TokyoClod in what way do the instructions in  [PX4 Developer Guide > Gazebo] (https://dev.px4.io/en/simulation/gazebo.html#start_px4_sim_separately) fail? These will need to be updated - see https://github.com/PX4/Devguide/issues/454

@EliaTarasov Thanks for that information. I can include it, but the existing material is centred around starting gazebo separately for debugging. Do you know how to do that (or can you just attach the IDE once you have started them separately as you indicated?)

EliaTarasov, 2018-02-09:
Yes i usually just attach debugger (Eclipse in my case).

closed 2018-05-10 by TSC21

cschindlbeck, 2020-08-19:
Can you be a little more verbose about the debugging process?

How can you attach a debugger in Eclipse?
Starting 
 `    make px4_sitl_default gazebo no_sim=1   `
in Eclipse shuts down immediately due to 
```
INFO  [px4] Startup script returned successfully
pxh> [2K
pxh> Shutting down
```

This is related to 
[https://discuss.px4.io/t/eclipse-debugging-sitl/9465](url)

TSC21, 2020-08-19:
@cschindlbeck please do not necrobump old tasks. Open a new issue and describe your setup, what you are trying to do, the problem and what you already tried to fix it.