PX4/PX4-SITL_gazebo-classic #116: Error is occurred when 'make' started

simkslv, 2017-08-01:
I download and install following instruction of Wiki.

But, at final, when I command "make" there was error as follows

make[2]: *** No rule to make target '../msgs/PROTOBUF_PROTOC_EXECUTABLE-NOTFOUND', needed by 'msgs/MotorSpeed.pb.cc'.  Stop.
CMakeFiles/Makefile2:450: recipe for target 'msgs/CMakeFiles/mav_msgs.dir/all' failed
make[1]: *** [msgs/CMakeFiles/mav_msgs.dir/all] Error 2
Makefile:83: recipe for target 'all' failed
make: *** [all] Error 2

After I got this error, I double check procedure and paths of configuration. But I couldn't find and fix it.

Please give me comments, Thank

dagar, 2017-08-02:
`PROTOBUF_PROTOC_EXECUTABLE-NOTFOUND`

Which protobuf packages do you have installed? Is it possible you first ran configure before all the dependencies were installed? Can you throw away the build directory and try again? Paste the output so we can see what was found.

simkslv, 2017-08-02:
Thank you for your comments.

After I got your comment, I delete "Build" and tried again. But I've got same result.

And I tried to re-install gazebo and dependencies. After that, I could success with make.

I think the problem is gazebo8, actually, I installed gazebo8 before. during I re-install, there was conflict with dependencies "libignition-math2"

Thank you again.

closed 2017-12-31 by TSC21