PX4/PX4-SITL_gazebo-classic #123: Unstability due to addition of camera

naveenT1010, 2017-09-08:
Hi, I am trying to add a camera on iris model to publish images on ROS. I am getting the image topic correctly but after adding the camera, the iris model has become very unstable. 

[image]

I think the added weight is giving an extra moment which is causing the oscillations, and in an attempt to stabilize the drone, the controller is overshooting. Can anyone suggest what would be a possible solution, should I change the PID values, or put the camera in the exact center?

anassinator, 2017-09-08:
I noticed the same thing for the plane model. Something I've done is simply setting the mass and inertia to the lowest settings by adding this to the camera model:

```
 <inertial>
   <mass>0.000001</mass>
   <inertia>
      <ixx>0.000001</ixx>
      <ixy>0.000000</ixy>
      <ixz>0.000000</ixz>
      <iyy>0.000001</iyy>
      <iyz>0.000000</iyz>
      <izz>0.000001</izz>
   </inertia>
</inertial>
```

naveenT1010, 2017-09-09:
@anassinator , i tried your solution, but it didn't work in my case. Did you put the camera at the center of the plane?

anassinator, 2017-09-09:
eyeballing it, it seems like it is near-ish center of gravity

naveenT1010, 2017-10-25:
It seems, after changing the moment of inertia matrix (setting it to minimum value possible), the effect of camera weight on the drone is now negligible, and it has become stable.

closed 2017-10-25 by naveenT1010