PX4/PX4-SITL_gazebo-classic #133: Gazebo installation failed

xdwgood, 2017-10-19:
Hi  all,
 I know it's not Px4's problem, but I really want De your help  
 I install it under the guidance of this http://gazebosim.org/tutorials page.
 When I run this command, "sudo apt-get install gazebo7", the error is as follows:
 
Reading package lists... Done
Building dependency tree       
Reading state information... Done
You might want to run 'apt-get -f install' to correct these:
The following packages have unmet dependencies:
 gazebo7 : Depends: libboost-program-options1.58.0 but it is not going to be installed
           Depends: libboost-regex1.58.0 but it is not going to be installed
           Depends: libboost-thread1.58.0 but it is not going to be installed
           Depends: libgazebo7 (= 7.8.1-1~xenial) but it is not going to be installed
           Depends: libsdformat4 but it is not going to be installed
           Depends: gazebo7-common (= 7.8.1-1~xenial) but it is not going to be installed
           Recommends: gazebo7-plugin-base
 linux-image-extra-4.10.0-35-generic : Depends: linux-image-4.10.0-35-generic but it is not going to be installed
 linux-image-extra-4.10.0-37-generic : Depends: linux-image-4.10.0-37-generic but it is not going to be installed
 linux-image-generic-hwe-16.04 : Depends: linux-image-4.10.0-37-generic but it is not going to be installed
 linux-signed-image-4.10.0-35-generic : Depends: linux-image-4.10.0-35-generic (= 4.10.0-35.39~16.04.1) but it is not going to be installed
 linux-signed-image-4.10.0-37-generic : Depends: linux-image-4.10.0-37-generic (= 4.10.0-37.41~16.04.1) but it is not going to be installed
E: Unmet dependencies. Try 'apt-get -f install' with no packages (or specify a solution).

**Then I enter this command "sudo apt-get -f install", and the error is as follows:**

dpkg: error processing archive /var/cache/apt/archives/linux-image-4.10.0-37-generic_4.10.0-37.41~16.04.1_amd64.deb (--unpack):
 cannot copy extracted data for './boot/vmlinuz-4.10.0-37-generic' to '/boot/vmlinuz-4.10.0-37-generic.dpkg-new': failed to write (No space left on device)
No apport report written because the error message indicates a disk full error
                                                                              dpkg-deb: error: subprocess paste was killed by signal (Broken pipe)
Examining /etc/kernel/postrm.d .
run-parts: executing /etc/kernel/postrm.d/initramfs-tools 4.10.0-37-generic /boot/vmlinuz-4.10.0-37-generic
run-parts: executing /etc/kernel/postrm.d/zz-update-grub 4.10.0-37-generic /boot/vmlinuz-4.10.0-37-generic
Preparing to unpack .../linux-image-4.10.0-35-generic_4.10.0-35.39~16.04.1_amd64.deb ...
Done.
Unpacking linux-image-4.10.0-35-generic (4.10.0-35.39~16.04.1) ...
dpkg: error processing archive /var/cache/apt/archives/linux-image-4.10.0-35-generic_4.10.0-35.39~16.04.1_amd64.deb (--unpack):
 cannot copy extracted data for './boot/abi-4.10.0-35-generic' to '/boot/abi-4.10.0-35-generic.dpkg-new': failed to write (No space left on device)
No apport report written because the error message indicates a disk full error
                                                                              dpkg-deb: error: subprocess paste was killed by signal (Broken pipe)
Examining /etc/kernel/postrm.d .
run-parts: executing /etc/kernel/postrm.d/initramfs-tools 4.10.0-35-generic /boot/vmlinuz-4.10.0-35-generic
run-parts: executing /etc/kernel/postrm.d/zz-update-grub 4.10.0-35-generic /boot/vmlinuz-4.10.0-35-generic
Errors were encountered while processing:
 /var/cache/apt/archives/linux-image-4.10.0-37-generic_4.10.0-37.41~16.04.1_amd64.deb
 /var/cache/apt/archives/linux-image-4.10.0-35-generic_4.10.0-35.39~16.04.1_amd64.deb
E: Sub-process /usr/bin/dpkg returned an error code (1)

Can someone help me? Thank you

xdwgood, 2017-10-19:
I'm using Ubuntu 16.04

tfoote, 2017-10-19:
This is not a gazebo issue either. You have run out of space on your `/boot` partition. 

```
dpkg: error processing archive /var/cache/apt/archives/linux-image-4.10.0-35-generic_4.10.0-35.3916.04.1_amd64.deb (--unpack):
cannot copy extracted data for './boot/abi-4.10.0-35-generic' to '/boot/abi-4.10.0-35-generic.dpkg-new': failed to write (No space left on device)
No apport report written because the error message indicates a disk full error
dpkg-deb: error: subprocess paste was killed by signal (Broken pipe)
```

You likely want to clear old kernels from there. A quick way to do that is `sudo apt-get autoremove`  but double check the list it will clean up does not include other important things before you let it run. If you have more issues please ask on an ubuntu forum.

closed 2017-10-19 by tfoote

xdwgood, 2017-10-20:
@tfoote   Thanks, I'm going to the forum to ask