PX4/PX4-SITL_gazebo-classic #152: CI: XML schema validation

TSC21, 2017-12-31:
We should add a way of validating the XML SDF schema files. Two options:
1. [lxml](http://lxml.de/)
2. [xmlsoft](http://xmlsoft.org/xmllint.html) - the most quickest way, as it is can be installed as a deb package.
To be included on a docker container.
@dagar what do you think?

dagar, 2017-12-31:
Do you want to enforce formatting as well?

TSC21, 2018-01-01:
> Do you want to enforce formatting as well?
We could I guess, but the most critical is the schema itself, like missing sign marks, brackets, others.

TSC21, 2018-01-01:
The schema for SDFormat files (SDF's)  seems to be this one http://sdformat.org/schemas/root.xsd.

TSC21, 2018-01-01:
Just an example of using the validator on the `solo.sdf` model, using the `model.xsd` schema (http://sdformat.org/schemas/root.xsd):
```bash
/home/tsc21/PX4_dev/src/sitl_gazebo/models/solo/solo.sdf:19: element pose: Schemas validity error : Element 'pose', attribute 'frame': The attribute 'frame' is not allowed.
/home/tsc21/PX4_dev/src/sitl_gazebo/models/solo/solo.sdf:57: element pose: Schemas validity error : Element 'pose', attribute 'frame': The attribute 'frame' is not allowed.
/home/tsc21/PX4_dev/src/sitl_gazebo/models/solo/solo.sdf:59: element pose: Schemas validity error : Element 'pose', attribute 'frame': The attribute 'frame' is not allowed.
/home/tsc21/PX4_dev/src/sitl_gazebo/models/solo/solo.sdf:90: element pose: Schemas validity error : Element 'pose', attribute 'frame': The attribute 'frame' is not allowed.
/home/tsc21/PX4_dev/src/sitl_gazebo/models/solo/solo.sdf:92: element pose: Schemas validity error : Element 'pose', attribute 'frame': The attribute 'frame' is not allowed.
/home/tsc21/PX4_dev/src/sitl_gazebo/models/solo/solo.sdf:104: element pose: Schemas validity error : Element 'pose', attribute 'frame': The attribute 'frame' is not allowed.
/home/tsc21/PX4_dev/src/sitl_gazebo/models/solo/solo.sdf:121: element pose: Schemas validity error : Element 'pose', attribute 'frame': The attribute 'frame' is not allowed.
/home/tsc21/PX4_dev/src/sitl_gazebo/models/solo/solo.sdf:156: element pose: Schemas validity error : Element 'pose', attribute 'frame': The attribute 'frame' is not allowed.
/home/tsc21/PX4_dev/src/sitl_gazebo/models/solo/solo.sdf:158: element pose: Schemas validity error : Element 'pose', attribute 'frame': The attribute 'frame' is not allowed.
/home/tsc21/PX4_dev/src/sitl_gazebo/models/solo/solo.sdf:170: element pose: Schemas validity error : Element 'pose', attribute 'frame': The attribute 'frame' is not allowed.
/home/tsc21/PX4_dev/src/sitl_gazebo/models/solo/solo.sdf:187: element pose: Schemas validity error : Element 'pose', attribute 'frame': The attribute 'frame' is not allowed.
/home/tsc21/PX4_dev/src/sitl_gazebo/models/solo/solo.sdf:222: element pose: Schemas validity error : Element 'pose', attribute 'frame': The attribute 'frame' is not allowed.
/home/tsc21/PX4_dev/src/sitl_gazebo/models/solo/solo.sdf:224: element pose: Schemas validity error : Element 'pose', attribute 'frame': The attribute 'frame' is not allowed.
/home/tsc21/PX4_dev/src/sitl_gazebo/models/solo/solo.sdf:236: element pose: Schemas validity error : Element 'pose', attribute 'frame': The attribute 'frame' is not allowed.
/home/tsc21/PX4_dev/src/sitl_gazebo/models/solo/solo.sdf:253: element pose: Schemas validity error : Element 'pose', attribute 'frame': The attribute 'frame' is not allowed.
/home/tsc21/PX4_dev/src/sitl_gazebo/models/solo/solo.sdf:288: element pose: Schemas validity error : Element 'pose', attribute 'frame': The attribute 'frame' is not allowed.
/home/tsc21/PX4_dev/src/sitl_gazebo/models/solo/solo.sdf:290: element pose: Schemas validity error : Element 'pose', attribute 'frame': The attribute 'frame' is not allowed.
/home/tsc21/PX4_dev/src/sitl_gazebo/models/solo/solo.sdf:302: element pose: Schemas validity error : Element 'pose', attribute 'frame': The attribute 'frame' is not allowed.
/home/tsc21/PX4_dev/src/sitl_gazebo/models/solo/solo.sdf:319: element pose: Schemas validity error : Element 'pose', attribute 'frame': The attribute 'frame' is not allowed.
/home/tsc21/PX4_dev/src/sitl_gazebo/models/solo/solo.sdf fails to validate
```

TSC21, 2018-01-01:
Removing the `frame=''` argument doesn't actually change nothing in terms of functionality, and the validator checks as good:
```bash
$ xmllint --format -schema http://sdformat.org/schemas/root.xsd /home/tsc21/PX4_dev/src/sitl_gazebo/models/solo/solo.sdf --noout
/home/tsc21/PX4_dev/src/sitl_gazebo/models/solo/solo.sdf validates
```

TSC21, 2018-01-15:
Solved with https://github.com/PX4/sitl_gazebo/pull/156

closed 2018-01-15 by TSC21