PX4/PX4-SITL_gazebo-classic #153: Build and Load Gazebo Plugin Library  Failed with Default CMakelist.txt

weiweikong, 2018-01-01:
## 1. Motivation
I tried to add a steer control plugin for the vehicle Polaris in `sitl_gazebo`, then we could control the Polaris and Quadrotor in Gazebo simulation environment simultaneously.

## 2. Some Configuration
### 2.1 Plugin Standalone Testing
This plugin has been tested standalone (https://github.com/weiweikong/polaris_simulation_standalone/blob/master/polaris_simple_control_plugin/src/polaris_simple_control_plugin.cpp) with ROS Kinect and Gazebo 7.

### 2.2 Add the plugin codes in `.sdf` file
```html
 <plugin name="polaris_simple_control_plugin" filename="libpolaris_simple_control_plugin.so">
      <alwaysOn>1</alwaysOn>
      <update>100</update>
      <updateRate>100.0</updateRate>
      <robotNamespace>polaris</robotNamespace>
      <leftJoint>rear_left_wheel_joint</leftJoint>
      <rightJoint>rear_right_wheel_joint</rightJoint>
      <leftsteerJoint>front_left_steering_joint</leftsteerJoint>
      <rightsteerJoint>front_right_steering_joint</rightsteerJoint>
      <wheelDistance>1.2</wheelDistance>
      <CarLength>1.88</CarLength>
      <wheelDiameter>0.32</wheelDiameter>
      <robotBaseFrame>base_link</robotBaseFrame>      
      <driveTorque>200</driveTorque>
      <steerTorque>200</steerTorque>
      <commandTopic>cmd_vel</commandTopic>      
      <broadcastTF>1</broadcastTF>
    </plugin>
```
## 2.3. Modify the `~/Firmware/Tools/sitl_gazebo/CMakeLists.txt` with
```
add_library(polaris_simple_control_plugin SHARED src/polaris_simple_control_plugin.cpp)
target_link_libraries(polaris_simple_control_plugin ${GAZEBO_LIBRARIES} ${catkin_LIBRARIES})
add_dependencies(polaris_simple_control_plugin mav_msgs)
```

## 2.4 Setup a new launch file, `polaris_empty.launch`
```html
<launch>
  <include file="$(find gazebo_ros)/launch/empty_world.launch">
    <arg name="paused" value="false"/>
    <arg name="use_sim_time" value="true"/>
    <arg name="gui" value="true"/>
    <arg name="headless" value="false"/>
    <arg name="debug" value="false"/>
  </include>px4
  <param name="robot_polaris" textfile="$(find px4)/Tools/sitl_gazebo/models/polaris/model.sdf" />
    <node
      name="spawn_model2"
      pkg="gazebo_ros"
      type="spawn_model"
      args="-sdf -param robot_polaris -model polaris -x 5.0"
      output="screen">
    </node>
</launch>
```
## 3. Errors
After building the plugin in `~/Firmware/Tools/sitl_gazebo/Build`
```
cmake .. && make
```
or in ``~/Fimware` with 
```
make posix_gazebo_sitl gazebo
```
I tried to run the launch file
```
roslaunch px4 polaris_empty.launch
```
and get the erros
```
...
 INFO] [1514777410.939556647, 0.158000000]: AckermannPlugin::Load
gzserver: symbol lookup error: ~/Firmware/build/posix_sitl_default/build_gazebo/libpolaris_simple_control_plugin.so: undefined symbol: _ZN2tf20TransformBroadcasterC1Ev
...
```
It seems the plugin had been loaded at first then crashed. In searching the internet, some developers gussed it, the symbol `_ZN2tf20TransformBroadcasterC1Ev`, might be the version confliction.

## 4. Some testing
- When I added the following codes in `../Firmware/Tools/sitl_gazebo/CMakeLists.txt`
```bash
find_package(catkin REQUIRED COMPONENTS roscpp rospy angles std_srvs geometry_msgs sensor_msgs nav_msgs urdf tf dynamic_reconfigure)
```
and run
```
cmake .. && make
```
at `../Firmware/Tools/sitl_gazebo/Build`
then I got the correct the library, though it located in `../Firmware/Tools/sitl_gazebo/Build/devel/lib`. -
- After copying the correct plugin, `libpolaris_simple_control_plugin.so` to the `~/Firmware/build/posix_sitl_default/build_gazebo`, the code run correctly.
- In my mind, it might be follow the `catkin_make` rules.

## 5. Questions
- What is the `_ZN2tf20TransformBroadcasterC1Ev` meaning ?
- What is the difference between the default `CMakefiles.txt` and the fixed one after add
```
find_package(catkin REQUIRED COMPONENTS roscpp rospy angles std_srvs geometry_msgs sensor_msgs nav_msgs urdf tf dynamic_reconfigure)
```
- What is the correct one to build the `catkin_make` based plugin in `sitl_gzebo` correctly?

TSC21, 2018-01-03:
> - What is the `_ZN2tf20TransformBroadcasterC1Ev` meaning ?

That's a symbolic link, and means you have a symbolic link error.

> - What is the difference between the default `CMakefiles.txt` and the fixed one after add

Fixed one? The difference is that `sitl_gazebo` does not have any kind of dependency of ROS, with exception of `gazebo_motor_failure_plugin`, which actually is just build if you use `-DBUILD_ROS_INTERFACE` cmake option. It also means it is required to be build with `catkin_make` or `catkin tools`.

> - What is the correct one to build the `catkin_make` based plugin in `sitl_gazebo` correctly?

`catkin_make` builds packages inside a catkin workspace, and usually it is used on the ROS environment. If you require ROS dependencies, it's advised to use `catkin_make` or `catkin tools`.

> - It seems that with the `catkin`, the libraries in `~/Tools/sitl_gazebo/Build/devel/lib` is correct, and could we set the `$GAZEBO_PLUGIN_PATH` to this folder?

Yes.

weiweikong, 2018-01-03:
@TSC21  Thank you for your kind reply.
So what is your suggestion to solve the symbolic link error, if I would like to build my gazebo plugin correctly by running `make posix_gazebo_sitl gazebo`, without copying the correct one into the `~/Firmware/build/posix_sitl_default/build_gazebo` folder.

TSC21, 2018-01-03:
`make posix_gazebo_sitl gazebo` uses raw `cmake` and not `catkin_make`, so you will have to make a choice or reconfigure how the CMakeLists looks for the dependencies.

TSC21, 2018-01-14:
@weiweikong is this solved to you?

weiweikong, 2018-01-15:
@TSC21 I just make the related library independently. So we could close this issue. If I solved all the erros, I will update this issue. Thank you for your help!

closed 2018-01-15 by weiweikong