PX4/PX4-SITL_gazebo-classic #159: Tf tree mismatch between sdf and rqt_tf_tree

margaridaCF, 2018-01-04:
Hi!
  I'm setting up the simulation for a uav equipped with a laser sensor. I'm experimenting with `roslaunch px4 posix_sitl.launch vehicle:=iris_rplidar est:=lpe`
  So far the configuration steps included below allow me to launch gazebo showing both UAV and sensor.
  In rviz, if I add the Laser Scan and switch the fixed frame to local_origin, I get
 ```
Transform [sender=unknown_publisher]
For frame [rplidar_link]: Frame [rplidar_link] does not exist
```
[image]

Why is the tree not showing the frames in the sitl_gazebo/models/iris_rplidar/iris_rplidar.sdf?
[image]

Thank you in advance.
Setup steps:
1. make posix_sitl_default gazebo
2. Kill process
1. source /ros_ws/devel/setup.bash
1. source Tools/setup_gazebo.bash /Firmware /Firmware/build/posix_sitl_default
1. export ROS_PACKAGE_PATH=$ROS_PACKAGE_PATH:/Firmware
1. export ROS_PACKAGE_PATH=$ROS_PACKAGE_PATH:/Firmware/Tools/sitl_gazebo
1. roslaunch px4 posix_sitl.launch vehicle:=iris_rplidar est:=lpe

TSC21, 2018-01-04:
Because there are no TF's being published from Gazebo side.

margaridaCF, 2018-01-04:
Thank you for the fast reply!
You mean I am missing the robot_state_publisher in the launch file, right?
I added to the posix_sitl.launch file:
```
<param name="robot_description" textfile="$(find mavlink_sitl_gazebo)/models/iris_rplidar/iris_rplidar.sdf" />
   <node pkg="robot_state_publisher" type="robot_state_publisher" name="robot_state_publisher">
        <param name="publish_frequency" type="double" value="30.0" />
    </node>
```
However it complains the sdf does not have a robot element.
This is a xacro vs urdf vs sdf issue, right?
I tried to use 
```
<param name="robot_description" command="$(find xacro)/xacro --inorder '$(find mavlink_sitl_gazebo)/models/rotors_description/urdf/iris_base.xacro'" />
```
instead but I get `Undefined substitution argument rotors_description_dir`
To me it seems that the make instructions (`make posix_sitl_default gazebo`) where the sdf for iris gets generated has some variables setup that are not visible while running roslaunch.

Which file should I load to robot_description?
Do you have any suggestion on how to make gazebo publish tf?

TSC21, 2018-01-04:
You are messing what is Gazebo and SDFs and what is ROS and URDFs. I advise you to check this for example: http://answers.gazebosim.org/question/4192/problem-with-sdf-files-or-robot_description-parameter/. You should probably start by creating a URDF for the Iris model and from there add the gazebo elements to make it work with both Gazebo and ROS world (Rviz, robot_description, robot_state_publisher).

margaridaCF, 2018-01-05:
Again thank you for your insightful comment (and patience lol)
I was indeed not getting what was what.

My progress so far didn't fix the tf. I get the same tree as before. 
I believe I am missing a source of JointState messages and haven't been able to localize one. I gather this from the robot_state_publisher [tutorial](http://wiki.ros.org/robot_state_publisher/Tutorials/Using%20the%20robot%20state%20publisher%20on%20your%20own%20robot).
I did a grep through the repo looking for `JointState` but couldn't find anything.
Is there something that can publish this messages? Am I completely on the wring track?

Here is my progress so far:
Generated an urdf from iris_base.xacro with the following commands:
```
python /Firmware/Tools/sitl_gazebo/scripts/xacro.py -o /Firmware/Tools/sitl_gazebo/models/rotors_description/urdf/iris_base.urdf /Firmware/Tools/sitl_gazebo/models/rotors_description/urdf/iris_base.xacro enable_mavlink_interface:=true enable_ground_truth:=false enable_wind:=false enable_logging:=false rotors_description_dir:=/Firmware/Tools/sitl_gazebo/models/rotors_description
sed -i 's/package:\/\/rotors_description/package:\/\/mavlink_sitl_gazebo\/models\/rotors_description/g' /Firmware/Tools/sitl_gazebo/models/rotors_description/urdf/iris_base.urdf
```
In the posix_sitl.launch file, after including the empty_world.launch and before closing the launch tag I have:
```
<param name="robot_description" textfile="/Firmware/Tools/sitl_gazebo/models/rotors_description/urdf/iris_base.urdf" />
<node pkg="robot_state_publisher" type="robot_state_publisher" name="robot_state_publisher">
    <param name="publish_frequency" type="double" value="30.0" />
</node>
<node name="$(anon vehicle_spawn)" output="screen" pkg="gazebo_ros" type="spawn_model"
     args="-urdf -param /robot_description -model iris 
     -x $(arg x) -y $(arg y) -z $(arg z) -R $(arg R) -P $(arg P) -Y $(arg Y)"/>
```

margaridaCF, 2018-01-08:
Finally figured out how to publish what seems to be the proper tf tree.
I was missing the joint_state_publisher. (I didn't know it was just another package). 
However I still cannot see the laser in rviz.
Do you understand what I'm doing wrong?

The full launch now looks like:
```
<launch>

    <!-- Posix SITL environment launch script -->
    <arg name="x" default="0"/>
    <arg name="y" default="0"/>
    <arg name="z" default="0"/>
    <arg name="R" default="0"/>
    <arg name="P" default="0"/>
    <arg name="Y" default="0"/>
    <arg name="est" default="lpe"/>
    <arg name="vehicle" default="iris"/>
    <arg name="world" default="$(find mavlink_sitl_gazebo)/worlds/empty.world"/>
    <arg name="sdf" default="$(find mavlink_sitl_gazebo)/models/$(arg vehicle)/$(arg vehicle).sdf"/>
    <arg name="rcS" default="$(find px4)/posix-configs/SITL/init/$(arg est)/$(arg vehicle)"/>

    <arg name="headless" default="false"/>
    <arg name="gui" default="true"/>
    <arg name="ns" default="/"/>

    <arg name="debug" default="false"/>
    <arg name="verbose" default="false"/>
    <arg name="paused" default="false"/>

    <node name="sitl" pkg="px4" type="px4" output="screen"
        args="$(find px4) $(arg rcS)">
    </node>

    <include file="$(find gazebo_ros)/launch/empty_world.launch">
        <arg name="headless" value="$(arg headless)"/>
        <arg name="gui" value="$(arg gui)"/>
        <arg name="world_name" value="$(arg world)" />
        <arg name="debug" value="$(arg debug)" />
        <arg name="verbose" value="$(arg verbose)" />
        <arg name="paused" value="$(arg paused)" />
    </include>

    <param name="robot_description" textfile="/Firmware/Tools/sitl_gazebo/models/rotors_description/urdf/iris_base.urdf" />
    <node pkg="robot_state_publisher" type="robot_state_publisher" name="robot_state_publisher">
        <param name="publish_frequency" type="double" value="30.0" />
    </node>
    <node name="joint_state_publisher" pkg="joint_state_publisher" type="joint_state_publisher" />
    <node name="$(anon vehicle_spawn)" output="screen" pkg="gazebo_ros" type="spawn_model"
        args="-urdf -param /robot_description -model iris -x $(arg x) -y $(arg y) -z $(arg z) -R $(arg R) -P $(arg P) -Y $(arg Y)"/>

</launch>
```
With this I can see:
[image]

However on rviz I can see nothing,
I the process I added a velodyne sensor on top and it can see nothing either.
Here is my Rviz

[image]

And the gazebo is:
[image]

TSC21, 2018-05-10:
@margaridaCF still a problem?

margaridaCF, 2018-05-11:
No, it's fixed.

TSC21, 2018-05-11:
OK good to know. Closing now.

closed 2018-05-11 by TSC21

Carbaton, 2018-09-01:
@margaridaCF How did you solve this problem? I've been trying to properly move the Tf tree in rviz while moving the drone in gazebo but i couldn't yet, any help will be appreciate.

Vrinda73, 2018-10-30:
Hello @margaridaCF ! I am trying to visualize the iris model in rviz and perform the offboard example. However I cannot see the model in rviz. It will be really helpful if you provide a starting point. Thanks much for your time on this !

margaridaCF, 2018-10-31:
Hi! Honestly, don't remember but you can look at the working version. I have it open sourced here https://github.com/margaridaCF/FlyingOctomap_code/tree/master/architecture/Tools 
I have a script to setup the model that contains the following instructions:
```
cp ~/Flying_Octomap_code/src/architecture/Tools/iris.xacro ~/src/Firmware/Tools/sitl_gazebo/models/rotors_description/urdf/
cp ~/Flying_Octomap_code/src/architecture/Tools/iris_base.xacro ~/src/Firmware/Tools/sitl_gazebo/models/rotors_description/urdf/
python ~/src/Firmware/Tools/sitl_gazebo/scripts/xacro.py -o ~/src/Firmware/Tools/sitl_gazebo/models/rotors_description/urdf/iris_base.urdf ~/src/Firmware/Tools/sitl_gazebo/models/rotors_description/urdf/iris_base.xacro enable_mavlink_interface:=true enable_ground_truth:=false enable_wind:=false enable_logging:=false rotors_description_dir:=/home/mfaria/src/Firmware/Tools/sitl_gazebo/models/rotors_description
sed -i 's/package:\/\/rotors_description/package:\/\/mavlink_sitl_gazebo\/models\/rotors_description/g' ~/src/Firmware/Tools/sitl_gazebo/models/rotors_description/urdf/iris_base.urdf
```
My directory tree has Flying_Octomap_code as a workspace in home and the src folder of px4 also in home.

Hope this helps.

saifullah3396, 2019-08-30:
I'm not sure if this is the right way with mavros but an example visualization for iris in rviz can be done simply with this kind of launch file. You might also need to turn map and base_link frame publishers in px4_config.yaml by setting send: true for the two frames.
```
<launch>
  <arg name="gui" default="true" />
  <arg name="rvizconfig" default="$(find ncai_uav_vis)/rviz/your_config.rviz" />
  <arg name="namespace" default="mavros" />
  <param name="robot_description" command="$(find xacro)/xacro '$(find rotors_description)/urdf/iris.xacro'" />
  <param name="use_gui" value="$(arg gui)"/>
  <node name="rviz" pkg="rviz" type="rviz" args="-d $(arg rvizconfig)" required="true" />
  <node name="robot_state_publisher" pkg="robot_state_publisher" type="state_publisher" />
  <node name="joint_state_publisher" pkg="joint_state_publisher" type="joint_state_publisher" />
</launch>
```

zahrakhan010, 2025-02-27:
I'm working with the X500 drone model in PX4 and using ROS2. Can you help me with the steps needed to visualize the drone model in RViz using an SDF file?