PX4/PX4-SITL_gazebo-classic #162: CMake fail, can not find _MAVLINK_INCLUDE_DIR

JcZou, 2018-01-09:
Hi

I am trying to build the drone model, but when I execute the cmake in sitl_gazebo/build,  the following error occurs. It seems can not find mavlink package. How should I install it and set MAVLINK_INCLUDE_DIRS, thanks!

-- Could NOT find MAVLink (missing:  MAVLINK_INCLUDE_DIRS) (found version "2.0")
-- catkin DISABLED
Gazebo version: 7.0
-- Boost version: 1.58.0
-- Found the following Boost libraries:
--   system
--   thread
--   timer
--   chrono
--   date_time
--   atomic
CMake Error: The following variables are used in this project, but they are set to NOTFOUND.
Please set them or make sure they are set and tested correctly in the CMake files:
_MAVLINK_INCLUDE_DIR

TSC21, 2018-01-14:
@JcZou do you have Mavlink headers on your system at all?

TSC21, 2018-01-15:
There are four ways you can have he headers on your system:
1. Clone the Mavlink repo and use pymavlink to generate the headers and put them in `/usr/local/include/` or something;
2. If you are using `sitl_gazebo` together with PX4 Fimware, that means you should already have the mavlink headers on your dirs, so the above error should not happen;
3. Clone https://github.com/mavlink/c_library_v2 headers and put them on a dir that can be found by `find_path()`, like `/usr/local/include/`;
4. Install the ROS Mavlink headers using `apt-get ros-<the_ros_dist_you_are_using>-mavlink`.

closed 2018-01-15 by TSC21

JcZou, 2018-01-16:
@TSC21  Thanks!

tokilligent, 2023-09-12:
@TSC21 I have installed both sitl_gazebo and PX4 firmware. But I am getting the same errors.

julianoes, 2025-03-13:
I think this is the actual solution: https://github.com/PX4/PX4-Autopilot/issues/20632#issuecomment-1483748872