PX4/PX4-SITL_gazebo-classic #170: MAVROS and GAZEBO - how to get actual position of the model

caomuqing, 2018-01-26:
Hi developers,

I am running simulation using Gazebo, and I use MAVROS to interface between ROS and gazebo to do some offboard mission. I would like to ask if there is a way/plugin/message that I can use to acquire the actual position of the simulated IRIS? I can get EKF estimated position from the /mavros/local_position/pose topic, but would want to have the actual position of the drone in the simulated environment, to do some comparison.

Thanks!

Best Regards

TSC21, 2018-01-26:
https://github.com/PX4/sitl_gazebo/blob/master/src/gazebo_mavlink_interface.cpp#L62

TokyoClod, 2018-04-06:
@TSC21 
I saw your answer for this question, but I found there's no  "/groundtruth" in my rostopic list after I run the px4 node with:
`roslaunch px4 mavros_posix_sitl.launch`
Can you show me a little more details to get the actual position?

Thanks for your effort!
Best Regards!

TSC21, 2018-04-06:
`/groundtruth` is not a rostopic. It is a gazebo topic and only used in gazebo internals. If you want to expose it to ROS, you need to use [gazebo_ros](http://wiki.ros.org/gazebo_ros).

closed 2018-04-06 by TSC21

TokyoClod, 2018-04-07:
@TSC21  It works after I added gazebo_ros_p3d plugin into iris.sdf as your mentioned. Thanks very much!
Could i ask a little more about the sitl_gazebo muddle?

I commented 
`addNoise(&linear_acceleration_I, &angular_velocity_I, dt)`
in (https://github.com/PX4/sitl_gazebo/blob/master/src/gazebo_imu_plugin.cpp#L305) 
as i did before in [https://github.com/PX4/Firmware/issues/8329](https://github.com/PX4/Firmware/issues/8329), trying to simulate the px4 in an ideal conditon without noise, in order to test a small vision algorithm. 
This method works on my old copy of Firmware cloned last year, but doesn't work on the up-to-date cloned copy.
May I ask you for some advice about this problem? It has troubled me quite a few days.
Thanks again!
Best Regards

TSC21, 2018-04-07:
https://github.com/PX4/sitl_gazebo/blob/master/models/rotors_description/urdf/iris_base.xacro#L49 change this to false (given that you are using the Iris model). Then recompile. (*Note:* You should not change the `irid.sdf` directly, since it is generated from xacro. If you want to make changes on the model, change the files on `rotors_description/urdf/` dir).