PX4/PX4-SITL_gazebo-classic #189: libgazebo_ros_camera Problem

Ahrovan, 2018-03-02:
@LorenzMeier sitl_gazebo has problem with run libgazebo_ros_camera, I followed the [(ros_interface)](https://dev.px4.io/en/simulation/ros_interface.html) for install and run gazebo simulation. Everything works fine, but after adding the camera, the following error occurs:

roslaunch px4 mavros_posix_sitl.launch
`
gzserver: /build/ogre-1.9-mqY1wq/ogre-1.9-1.9.0+dfsg1/OgreMain/src/OgreRenderSystem.cpp:546: virtual void Ogre::RenderSystem::setDepthBufferFor(Ogre::RenderTarget*): Assertion `bAttached && "A new DepthBuffer for a RenderTarget was created, but after creation" "it says it's incompatible with that RT"' failed.
Aborted
[gazebo-3] process has died [pid 20750, exit code 134, cmd /opt/ros/kinetic/lib/gazebo_ros/gzserver -e ode /home/cv/src/Firmware/Tools/sitl_gazebo/worlds/empty.world __name:=gazebo __log:=/home/cv/.ros/log/68ef15b6-1e60-11e8-b0c8-000c29ddf392/gazebo-3.log].
log file: /home/cv/.ros/log/68ef15b6-1e60-11e8-b0c8-000c29ddf392/gazebo-3*.log
`

this code added to iris.sdf:
`
      <joint name='DownCam_joint' type='fixed'>
        <child>camera_link</child>
        <parent>base_link</parent>
      </joint>
      <link name='camera_link'>
        <pose frame=''>0.1 0 0.194923 0 0 0</pose>
        <inertial>
          <pose frame=''>0 0 0 0 0 0</pose>
          <mass>1e-05</mass>
          <inertia>
            <ixx>0.0001</ixx>
            <ixy>0</ixy>
            <ixz>0</ixz>
            <iyy>0.0002</iyy>
            <iyz>0</iyz>
            <izz>0.0002</izz>
          </inertia>
        </inertial>
        <visual name='visual'>
          <geometry>
            <box>
              <size>0.005 0.005 0.005</size>
            </box>
          </geometry>
        </visual>
        <sensor name='FrontCam' type='camera'>
          <update_rate>30</update_rate>
          <camera name='head'>
            <horizontal_fov>1.39626</horizontal_fov>
            <image>
              <width>720</width>
              <height>480</height>
              <format>R8G8B8</format>
            </image>
            <clip>
              <near>0.02</near>
              <far>300</far>
            </clip>
            <noise>
              <type>gaussian</type>
              <mean>0</mean>
              <stddev>0.007</stddev>
            </noise>
          </camera>
          <plugin name='camera_controller' filename='libgazebo_ros_camera.so'>
            <alwaysOn>true</alwaysOn>
            <updateRate>0.0</updateRate>
            <cameraName>birrobotic/FrontCam</cameraName>
            <imageTopicName>image_raw</imageTopicName>
            <cameraInfoTopicName>camera_info</cameraInfoTopicName>
            <frameName>camera_link</frameName>
            <hackBaseline>0.07</hackBaseline>
            <distortionK1>0.0</distortionK1>
            <distortionK2>0.0</distortionK2>
            <distortionK3>0.0</distortionK3>
            <distortionT1>0.0</distortionT1>
            <distortionT2>0.0</distortionT2>
          </plugin>
        </sensor>
        <self_collide>0</self_collide>
        <enable_wind>0</enable_wind>
        <kinematic>0</kinematic>
      </link>
`

The same problem has already been addressed and debugged in another platform [(osrf)](https://github.com/osrf/uctf/issues/116)

Ahrovan, 2018-03-03:
@All How to install libgazebo_ros_camera for sitl_gazebo to publish camera images?

Ahrovan, 2018-03-23:
solved.

closed 2018-03-23 by TSC21

sanjainn, 2021-02-26:
Hi @Ahrovan , I literally have the same issue as you in https://github.com/osrf/uctf/issues/116.
But I have Gazebo 9 with Ros kinetic i Ubuntu 16 and no PX4 installed. I am only using the arducopter gazebo package to work on getting the rostopics of the camera plugin, but not able to. I have been able to attach the camera to the iris model by adding the camera sdf files in the world file, but without the plugin. Because every time I use the plugin in the world or in the model.sdf file of either the camera or the iris model, I get error while launching like this:

[gazebo-1] process has died [pid 22317, exit code 139, cmd /opt/ros/kinetic/lib/gazebo_ros/gzserver -e ode worlds/sagar_defense.world __name:=gazebo __log:=/home/sanil/.ros/log/471ace3a-77ea-11eb-8b7d-5800e3d1b0ef/gazebo-1.log].
log file: /home/sanil/.ros/log/471ace3a-77ea-11eb-8b7d-5800e3d1b0ef/gazebo-1*.log

Can you pls help in letting me know how to create the rostopics of the camera plugin in the drone?? I am tired of being confused here!!!