PX4/PX4-SITL_gazebo-classic #192: Can not connect to Intel AERO FC in HITL mode via UDP

trgiman, 2018-03-23:
Hi all. Currently I am trying to use Gazebo to simulate Intel Aero FC in HITL mode. But without success. (I have been successful in connecting jMavSim after modifications. https://github.com/PX4/jMAVSim/issues/74 ). 

Intel AERO RTF is using UDP connection and mavlink-router trough it's USB port on Aero Board.But after looking on

if (serial_enabled_) {
    // gcs link
    _myaddr.sin_addr.s_addr = mavlink_addr_;
    _myaddr.sin_port = htons(mavlink_udp_port_);
    _srcaddr.sin_addr.s_addr = qgc_addr_;
    _srcaddr.sin_port = htons(qgc_udp_port_);
  }

  else {
    _myaddr.sin_addr.s_addr = htonl(INADDR_ANY);
    // Let the OS pick the port
    _myaddr.sin_port = htons(0);
    _srcaddr.sin_addr.s_addr = mavlink_addr_;
    _srcaddr.sin_port = htons(mavlink_udp_port_);
  }

in https://github.com/PX4/sitl_gazebo/blob/master/src/gazebo_mavlink_interface.cpp

Problem is, that I need to connect gazebo plugin to exact remote ip address of AeroFC(192.168.7.2) and port of mavlink router (which is always different, you can see all configs from issue link above)

I think that according code above I can use only serial port and serial port will be too slow.Any suggestions please?

TSC21, 2018-03-23:
Currently Gazebo only supports HITL through serial port, using UDP to connect to the GCS. You can add support for having a secondary UDP port for connecting to the remote port, but you have to do some modifications in the logic above. A PR is welcomed and me or @arwagoner can review it after.

trgiman, 2018-03-27:
@TSC21 I did quick modification to be able to test Gazebo. Result is that AeroFC receives simulated data. I can control drone (Iris model) trough RC or trough QGC. But something is wrong....since it is very hard to fly,take-off Iris or other models in position,altitude any modes. Can you please suggest? Some configuration within sdf files? I must be missing something essential(also when taking off it seems rotors on one side are getting higher PWM, than those on the other side)

**On Aero Board (Yocto):**

Configuration of mavlink-router is (14550 for QGC, 14560 for sitl_gazebo):

[General]
TcpServerPort=0
ReportStats=false
Log=/var/lib/mavlink-router/
****
[UartEndpoint uart]
Device = /dev/ttyS1
Baud = 921600,460800

[UdpEndpoint ethernet-over-usb]
Mode = Normal
Address = 192.168.7.255
Port = 14560

[UdpEndpoint ethernet-over-usb]
Mode = Normal
Address = 192.168.7.255
Port = 14550

**On my Ubuntu laptop within sitl_gazebo(for now only quick hack):**
if (serial_enabled_) {
// gcs link
myaddr.sin_addr.s_addr = mavlink_addr;
myaddr.sin_port = htons(mavlink_udp_port);
srcaddr.sin_addr.s_addr = qgc_addr;
srcaddr.sin_port = htons(qgc_udp_port);
}

else {
//_myaddr.sin_addr.s_addr = htonl(INADDR_ANY);
// Let the OS pick the port
//_myaddr.sin_port = htons(0);
// srcaddr.sin_addr.s_addr = mavlink_addr;
// srcaddr.sin_port = htons(mavlink_udp_port);
myaddr.sin_addr.s_addr = mavlink_addr;
myaddr.sin_port = htons(mavlink_udp_port);
srcaddr.sin_addr.s_addr = qgc_addr;
srcaddr.sin_port = htons(qgc_udp_port);
}

**In iris.sdf:**

serial disabled
hitl enabled

<robotNamespace/>
      <imuSubTopic>/imu</imuSubTopic>
      <gpsSubTopic>/gps</gpsSubTopic>
      <mavlink_addr>INADDR_ANY</mavlink_addr>
      <mavlink_udp_port>14560</mavlink_udp_port>
      <serialEnabled>false</serialEnabled>
      <serialDevice>/dev/ttyACM0</serialDevice>
      <baudRate>921600</baudRate>
      <qgc_addr>INADDR_ANY</qgc_addr>
      <qgc_udp_port>14550</qgc_udp_port>
      <hil_mode>true</hil_mode>
      <hil_state_level>false</hil_state_level>

**On Aero FC:**

1.HITL enabled
2.tap_esc not running

**Sitl_gazebo:**

Everytime when drone crash I need to click Reset Model Pose. After this action PX4 Aero  FC console strats produce following messages (baro related and GPS related):

WARN  [lib__ecl] EKF baro hgt timeout - reset to baro
WARN  [lib__ecl] EKF baro hgt timeout - reset to baro

trgiman, 2018-03-27:
@TSC21 another update:

When in nsh console and run listener actuator_update command. Sometimes PX4 goes from 4 to 6 outputs?

TOPIC: actuator_outputs instance 0 #1971
timestamp: 879450160
noutputs: 6
output: 2000.0000 1000.0000 1538.0183 1538.0183 1500.0000 1500.0000      NaN      NaN      NaN      NaN      NaN      NaN      NaN      NaN      NaN      NaN

TOPIC: actuator_outputs instance 0 #1972
timestamp: 879491012
noutputs: 4
output: 1190.0000 1190.0000 1190.0000 1190.0000      NaN      NaN      NaN      NaN      NaN      NaN      NaN      NaN      NaN      NaN      NaN      NaN

TOPIC: actuator_outputs instance 0 #1973
timestamp: 879533012
noutputs: 4
output: 1190.0000 1190.0000 1190.0000 1190.0000      NaN      NaN      NaN      NaN      NaN      NaN      NaN      NaN      NaN      NaN      NaN      NaN

TOPIC: actuator_outputs instance 0 #1974
timestamp: 879571709
noutputs: 6
output: 2000.0000 1000.0000 1537.7014 1537.7014 1500.0000 1500.0000      NaN      NaN      NaN      NaN      NaN      NaN      NaN      NaN      NaN      NaN

Is it not wrong configuration of iris.sdf model?

arwagoner, 2018-03-27:
@trgiman, the problem isn't the configuration of iris.sdf model. The issue is that mavlink-router drops a significant enough portion of packets being sent and received that causes inconsistent actuator_output commands. Is there any reason you can't use a serial connection?

trgiman, 2018-03-27:
@arwagoner Intel RTF uses serial port  but it is not fast enough? Only 57800 bauds-112 bauds. Usb port is only on AeroBoard and that one is used for ethernet over usb as per my config above. In case of jMavsim sim actuators were working correctly. There were no issues and jMavSim was communicating trough mavlink-router as well. Could it be something else?

trgiman, 2018-03-28:
Hi @TSC21 @arwagoner  I do have an progress. I was able to reduce sequence lost rate for UART 5 to reasonable levels by removing Log from mavlink-router main.conf. It helped with jMavsim, but not to sitl_gazebo. 

1. For sitl_gazebo UDP4 reports like 66% sequence lost (see log below)
2. USART 5 (routing from AeroBoard to AeroFC) is max upto 20%
3.Any ideas why such a lost on UDP4 for sitl_gazebo?
4. In jMavsim I use -r parameter (it is update rate for MAV messages). Can update rate be an problem for sitl_gazebo --> mavlink router ---> Aero FC? 

**1. jMavsim log after removing log from mavlink-router**

Endpoint UDP [4] {
	Received messages {
		CRC error: 0 0% 0KBytes
		Sequence lost: 0 0%
		Handled: 41528 3064KBytes
		Total: 41528
	}
	Transmitted messages {
		Total: 96291 4544KBytes
	}
}
Endpoint UART [5] {
	Received messages {
		CRC error: 34 0% 1KBytes
		Sequence lost: 16051 14%
		Handled: 96291 4544KBytes
		Total: 112376
	}

**2. sitl-gazebo mavlink-router log after removing log from mavlink-router main.conf** 

Endpoint UDP [4] {
	Received messages {
		CRC error: 0 0% 0KBytes
		Sequence lost: 33583 66%
		Handled: 17155 1225KBytes
		Total: 50738
	}
	Transmitted messages {
		Total: 108395 5164KBytes
	}
}
Endpoint UART [5] {
	Received messages {
		CRC error: 11 0% 0KBytes
		Sequence lost: 3092 2%
		Handled: 108395 5164KBytes
		Total: 111498
	}
	Transmitted messages {
		Total: 17169 1226KBytes
	}

trgiman, 2018-03-28:
@arwagoner I decreased world update rate in Gazebo on 250Hz. Error rate and packet lost dropped down rapidly 20% on UDP4 and 2% on UART5. But it is strange. There should be no packet lost on UDP4?Could it be SYS_ID conflict?

If I will opt for connecting communication trough RXTX UART port. Is 115200 bauds enough for communication? 

I do not understand why jMavsim works correctly and sitl_gazebo plugin is not trough UDP? Could you please help to understand or suggest some hints in code?

trgiman, 2018-03-30:
Hi @julianoes @bkueng sorry for disturbing,  few days ago we have been discussing few jMavsim issues, which were solved. I am struggling now to understand to compare some parts of sitl_gazebo code and jMavsim due to sitl_gazebo HITL setup issues.

1. jMavsim in HITL mode trough Intel Mavlink Router to Aero FC works correctly (there is low sequence lost on UART5 as mentioned above 5%, but that is ok. I improved it also by switching off mavlink log in mavlinkrouter)

2.sitl_Gazebo is not usable (comments above). Since immediately when I connect to Intel Mavlink Router. **UDP 4** sequence lost is 66%. **Why UDP endpoint?** I have been trying to compare code in between on jMavsim and sitl_gazebo, but can not find the reason why sitl_gazebo is not working for HITL setup (messages are getting lost...drone is therefore uncontrollable. SITL mode with sitl_gazebo running on one machine works great). jMavsim works great in HITL.

a) checked compared system_id, component_id in formed Mavlink messages 
b) checked compared UDP socket creating mechanism

Could you please help me to orient a bit more?

TSC21, 2018-03-30:
@trgiman the implementation of HITL to jMAVSim and to sitl_gazebo had major differences as they are completely different simulators. HITL for sitl_gazebo works great if you actually use the serial interface instead of UDP. My first try to connect sitl_gazebo with HITL through mavlink-router to an autopilot was also a failure given to high package drop and that's why me and @arwagoner moved to implement the serial interface instead. So the conclusion of this is that the problem maybe be in the skew introduced by mavlink-router.

TSC21, 2018-03-30:
There maybe also some differences on the UDP interface that may cause jMAVSim to work "faster" with mavlink-router but I can't actually debug this atm.

trgiman, 2018-03-30:
@TSC21 so even if serial port of AeroRTF can do max 115200 bauds. It is still viable solution? For instance for jMavsim @bkueng mentioned, that it would not be enough. Ok. I will try serial version again. Anyway thanks a lot for inputs

bkueng, 2018-04-03:
@trgiman it might be that gazebo does not check for mavlink messages fast enough on the UDP port. Either by a code design problem or just that gazebo requires more CPU and you hit that limit on your machine.

trgiman, 2018-04-03:
@bkueng > just that gazebo requires more CPU and you hit that limit on your machine.

1.I can run SITL with gazebo without any issues on my computer. Simulation works without any issues.

2. Intel TELEM port 115200 baud is too slow for using sitl_gazebo in serial mode. I tried again. Messages are not coming fast enough...

3. Switched off mavlink-routerd and used mavlink_bridge.py - it works for JMavsim but not for sitl_gazebo plugin.

4. Does it makes sense to try RotorS with hil interface to try to make it work?....as I can see, that sitl_gazebo code and RotorS hil interface are very similar....also still can not understand why when I connect with sitl_gazebo - sequence lost on UDP4 interface is 66%

Anyway thanks for all advices, ideas.

TSC21, 2018-05-31:
@trgiman what's the status on this?

trgiman, 2018-05-31:
@TSC21 I was not able to solve the issue with Gazebo. It worked with jMavsim however (point 3 above).Unfortunately It seems I am limited with Intel RTF setup serial baud rate (unfortunately I can use  only buildin FC of Aero I do not have got external FC like Pixhawk). I had to abandon these HITL mode kind of simulations for my case for now.

TSC21, 2018-05-31:
> Switched off mavlink-routerd and used mavlink_bridge.py - it works for JMavsim but not for sitl_gazebo plugin.

What `mavlink_bridge.py` is this?

TSC21, 2018-05-31:
@anitha-suresh is there any recommend setup from your side for testing Aero on simulation? Have you tried to setup HITL with it? Do you have any suggestions for having working properly with Gazebo? Thanks

julianoes, 2018-06-01:
@bkueng @trgiman last time I checked the mavlink message rate set for SITL were set way too fast, basically for the purpose of something like a VIO integration, when really much lower rates would completely do for SITL.

lbegani, 2018-06-01:
> @anitha-suresh is there any recommended setup from your side for testing Aero on simulation? Have you tried to setup HITL with it? Do you have any suggestions for having working properly with Gazebo? Thanks

Have not tried such setup, unfortunately cannot help.

trgiman, 2018-06-01:
hi @julianoes basically as described above I also tried to decrease message update rate in gazebo sitl plugin. I was using it for HITL mode. For now my result is that only HITL setup with stock Intel Aero RTF with build in FC is by using only JMavsim. Baud rate of serial port of Intel RTF for Gazebo is too slow even when I disabled logging mavlink messages to Intel Aero Board etc.

@TSC21 I was using this  mavlink_bridge.py(https://gist.githubusercontent.com/mindThomas/51f10f9f2c1d9aa2650adce55bc001f1/raw/5acd50c38c8dd53d3eaa001c759aL) but independently of using mavlink-routrd or mavlink_bridge JMavsim worked and SITL_GAZEBO not.

TSC21, 2018-06-01:
> @bkueng @trgiman last time I checked the mavlink message rate set for SITL were set way too fast, basically for the purpose of something like a VIO integration, when really much lower rates would completely do for SITL.

@julianoes we can always throttle the msg pub on the `gazebo_mavlink_interface`.

TSC21, 2018-06-01:
> @TSC21 I was using this mavlink_bridge.py(https://gist.githubusercontent.com/mindThomas/51f10f9f2c1d9aa2650adce55bc001f1/raw/5acd50c38c8dd53d3eaa001c759aL) but independently of using mavlink-routrd or mavlink_bridge JMavsim worked and SITL_GAZEBO not.

This may be related with the way the serial interface is implemented on the Gazebo. This can be a review target later. Let's leave this issue opened so it can be tracked.

julianoes, 2018-06-04:
> @julianoes we can always throttle the msg pub on the gazebo_mavlink_interface.

Ok my point was rather that the default is too fast. It would make more sense to start slow and have users of the messages request faster rates if needed.

closed 2020-04-25 by LorenzMeier