PX4/PX4-SITL_gazebo-classic #193: Vision plugin correct?

julianoes, 2018-03-24:
I just stumbled upon this:

https://github.com/PX4/sitl_gazebo/blob/90c9e5db3f53b0d963a4f240cc712f2ecf246bc4/src/gazebo_vision_plugin.cpp#L133-L135

and was wondering if roll and pitch are swapped on purpose?
[gazebo::math::Quaternion::SetFromEuler](http://osrf-distributions.s3.amazonaws.com/gazebo/api/4.0.0/classgazebo_1_1math_1_1Quaternion.html#aed29d19853fde3fc2662974c44811346) has the ordering roll, pitch yaw.

@ChristophTobler @mrivi

TSC21, 2018-03-24:
@julianoes I think that may be a typo.

TSC21, 2018-04-30:
https://ignitionrobotics.org/api/math/0.1/classignition_1_1math_1_1Quaternion.html#a1ad9bd946d762ce60994979cec1bfd4e - `SetFromEuler()` has `RPY` order, so this is a type. I find awkward that the estimators don't complain though.

TSC21, 2018-04-30:
> I find awkward that the estimators don't complain though.

Well the reason is because none of the PX4 estimators fuses roll or pitch from vision, just yaw. So that's why it never complaint.