PX4/PX4-SITL_gazebo-classic #204: gazebo_mavlink_interface.cpp: error

mashenlyl, 2018-05-06:
/home/lyl/PX4/src/Firmware/Tools/sitl_gazebo/src/gazebo_mavlink_interface.cpp: In member function ‘void gazebo::GazeboMavlinkInterface::IRLockCallback(gazebo::IRLockPtr&)’:
/home/lyl/PX4/src/Firmware/Tools/sitl_gazebo/src/gazebo_mavlink_interface.cpp:761:14: error: ‘mavlink_landing_target_t’ has no member named ‘position_valid’
   sensor_msg.position_valid = false;
              ^
/home/lyl/PX4/src/Firmware/Tools/sitl_gazebo/src/gazebo_mavlink_interface.cpp:762:14: error: ‘mavlink_landing_target_t’ has no member named ‘type’
   sensor_msg.type = LANDING_TARGET_TYPE_LIGHT_BEACON;
              ^
/home/lyl/PX4/src/Firmware/Tools/sitl_gazebo/src/gazebo_mavlink_interface.cpp:762:21: error: ‘LANDING_TARGET_TYPE_LIGHT_BEACON’ was not declared in this scope
   sensor_msg.type = LANDING_TARGET_TYPE_LIGHT_BEACON;
                     ^
/home/lyl/PX4/src/Firmware/Tools/sitl_gazebo/src/gazebo_mavlink_interface.cpp: In member function ‘void gazebo::GazeboMavlinkInterface::handle_message(mavlink_message_t*)’:
/home/lyl/PX4/src/Firmware/Tools/sitl_gazebo/src/gazebo_mavlink_interface.cpp:839:8: error: ‘MAVLINK_MSG_ID_HIL_ACTUATOR_CONTROLS’ was not declared in this scope
   case MAVLINK_MSG_ID_HIL_ACTUATOR_CONTROLS:
        ^
/home/lyl/PX4/src/Firmware/Tools/sitl_gazebo/src/gazebo_mavlink_interface.cpp:840:5: error: ‘mavlink_hil_actuator_controls_t’ was not declared in this scope
     mavlink_hil_actuator_controls_t controls;
     ^
/home/lyl/PX4/src/Firmware/Tools/sitl_gazebo/src/gazebo_mavlink_interface.cpp:840:37: error: expected ‘;’ before ‘controls’
     mavlink_hil_actuator_controls_t controls;
                                     ^
/home/lyl/PX4/src/Firmware/Tools/sitl_gazebo/src/gazebo_mavlink_interface.cpp:841:52: error: ‘controls’ was not declared in this scope
     mavlink_msg_hil_actuator_controls_decode(msg, &controls);
                                                    ^
/home/lyl/PX4/src/Firmware/Tools/sitl_gazebo/src/gazebo_mavlink_interface.cpp:841:60: error: ‘mavlink_msg_hil_actuator_controls_decode’ was not declared in this scope
     mavlink_msg_hil_actuator_controls_decode(msg, &controls);

TSC21, 2018-05-06:
Your Mavlink version is outdated. How are you building this? Inside or outside the PX4 Firmware?

mashenlyl, 2018-05-08:
@TSC21  Inside the PX4 Firmware.

TSC21, 2018-05-08:
Do you have your mavlink submodule up to date?

mashenlyl, 2018-05-08:
Yes. Mavlink v2.0

TSC21, 2018-05-08:
No, not that. The git tag version of the submodule branch.

mashenlyl, 2018-05-09:
the branch is 748192f

TSC21, 2018-05-10:
OK. Do you actually have ROS Mavlink installed on your system?

mashenlyl, 2018-05-11:
I remove the ros-indigo-mavros package and solve it. Thank You!

closed 2018-05-11 by mashenlyl

TSC21, 2018-05-11:
This commit https://github.com/PX4/sitl_gazebo/commit/0298607833ba78c001eda723e3da0450bfcadbb7 solves this issue. You can reinstall `ros-indigo-mavros` and respective dependencies without any problem.