PX4/PX4-SITL_gazebo-classic #205: Need Octocopter_X Model

TobiMiller, 2018-05-08:
Hey,

i want to create a DJI S1000 model. I have created it but it doesnt fly like expected. Is there any sample model working as octocopter? I can provide the ROS package if needed.

Best regards

TSC21, 2018-05-10:
On this repo you won't find an octocopter model. I advise you to take a look on how the common `iris` model or the solo is generated and, from there, add new rotors to it, the respective Mavlink actuators and, in the PX4 side, use the `8001_octo_x` config file for example.
The ROS package won't help much, but providing the SDF would help us debug it.

TobiMiller, 2018-05-14:
Hey @TSC21 ,

sorry for this late answer. I have used the Typhoon as example and changed it for my needs. Here you go. The motor constants are calculated by some equations you wrote in this issue board. i think my calculation is bullshit. the standard values doesnt work very well also.

[dji_s1000.zip](https://github.com/PX4/sitl_gazebo/files/1999648/dji_s1000.zip)

TSC21, 2018-05-14:
I would advise you not to play with the motor constants and use the default motors. The equations you found in https://github.com/PX4/sitl_gazebo/issues/110 are still yet to confirm.

TobiMiller, 2018-05-15:
ok fine. The behaviour with the default motor constants is also unsatisfying.

TSC21, 2018-05-31:
@TobiMiller you should look on optimizing the parameters for the motor models and even play with the PID tuning for the controllers. Also, without some logs to explain and analyse the that "unsatisfying" behavior it becomes difficult to help you solve the problem.
Also I advise you to instead of sharing a zip file, you create a branch and make a PR of your code so we can analyse your model.

TobiMiller, 2018-06-08:
@TSC21 & @LorenzMeier 

will give it a try in july. until then this topic is on hold.

chrismath, 2018-08-20:
I want to add the astec firefly model i have the xacro file how do i add it

TSC21, 2018-11-19:
Closing as stale. @chrismath please open a new issue if you want help solving your problem.

closed 2018-11-19 by TSC21