PX4/PX4-SITL_gazebo-classic #228: how to enable camera and take its feed

chrismath, 2018-08-20:
I have been trying to enable camera in this simulator.I did make some changes in the cmake file but the there is no topic providing the camera output

EliaTarasov, 2018-08-20:
Hey chrismath!
1. What vehicle do you actually test?
2. What camera do you want to use? (e.g. optical flow cam, vision cam, gimbal cam).

In general sitl_gazebo can output images to gazebo cam topic and to ros cam topic. So you may use `gz topic -l` for gazebo topics or `rostopic list` for ros topics.

chrismath, 2018-08-21:
I was planning to do indoor navigation,  I have been working with the iris quadrotor in the simulator. I want to use vision cam. But on running rostopic list there are no topics related to image_raw

chrismath, 2018-08-24:
Thank you EliaTarasov ill try it out.Sorry for the late reply

EliaTarasov, 2018-09-21:
@chrismath any news on this issue?

chrismath, 2018-10-02:
nope

EliaTarasov, 2018-10-03:
Did you try and it didn't work out?

Veilkrand, 2018-12-02:
I launch `px4 mavros_posix_sitl.launch vehicle:=iris_fpv_cam` from another launch file and the model loads and shows at Gazebo. However PX4 complains at start:
```
px4 starting.

INFO  [px4] Calling startup script: /bin/sh etc/init.d-posix/rcS 0
Error: Unknown model 'iris_fpv_cam'
etc/init.d-posix/rcS: 48: exit: Illegal number: -1
ERROR [px4] Startup script returned with return value: 512
```
What am I missing? Thanks!

EliaTarasov, 2018-12-02:
@Veilkrand this can happen if you external launch file does not take into account that `roslaunch px4 mavros_posix_sitl.launch` command must be executed from `src/Firmware` directory.

Veilkrand, 2018-12-02:
First I setup the globals in here:
```
#!/bin/bash
px4_dir=~/Firmware
#source $px4_dir/Tools/setup_gazebo.bash $px4_dir $px4_dir/build/posix_sitl_def$
source $px4_dir/Tools/setup_gazebo.bash $px4_dir $px4_dir/build/px4_sitl_default
export ROS_PACKAGE_PATH=$ROS_PACKAGE_PATH:$px4_dir
export ROS_PACKAGE_PATH=$ROS_PACKAGE_PATH:$px4_dir/Tools/sitl_gazebo
```

And then I use this launch file:
```
<?xml version="1.0"?>
<launch>
  <include file="$(find px4)/launch/mavros_posix_sitl.launch">
    <arg name="vehicle" value="iris_fpv_cam" />
  </include>
  <node name="offb" pkg="offb" type="offb_node" output="screen"/>
</launch>
```
It finds the px4 package and the model. What am I doing wrong?

Veilkrand, 2018-12-03:
I think I got what was the problem. If you change: `<arg name="vehicle" value="iris_fpv_cam" />` Gazebo will load the right .sdf model but PX4 won't find the target with the name of the model. So you need to keep the `name` argument as `iris`, and change directly the `sdf` argument.
```
<arg name="sdf" value="$(find mavlink_sitl_gazebo)/models/$(arg my_model_name)/$(arg my_model_name).sdf"/>
```

EliaTarasov, 2018-12-03:
Great! So the problem solved?

Veilkrand, 2018-12-03:
It works for me. But I wonder why `roslaunch px4 mavros_posix_sitl.launch vehicle:=iris_fpv_cam` works for you. I should be missing something

nkhedekar, 2019-05-30:
Following the instructions mentioned [here](https://github.com/PX4/Devguide/issues/66#issue-191233954) and 

> I think I got what was the problem. If you change: `<arg name="vehicle" value="iris_fpv_cam" />` Gazebo will load the right .sdf model but PX4 won't find the target with the name of the model. So [...]

it worked perfectly.

closed 2020-04-25 by LorenzMeier

AlexWUrobot, 2023-01-19:
I solve this by this ref link: https://ubungmachtdenmeistersite.wordpress.com/2022/11/23/tracking-ar-tags-in-ros/ 

If you want to install USB camera on Ubuntu 18.04, ROS Melodic, and OpenCV python 2.7

I search so many websites and find this two website is very useful.

    https://varunagrawal.github.io/2017/10/23/ar-track/
    http://projectsfromtech.blogspot.com/2017/09/tracking-ar-tags-with-ros-monocular.html

However, I follow the instruction of this two websites, but something cannot work.

In the end, I adjust the tracker.launch file like this below screenshot. Then I succeed in my drone onboard PC, Jetson TX1.

I also find you have to notice, if you using Ubuntu 20.04, the default python version is 3.6.

However, the ROS AR tracking code uses the python version is 2.7.

You may need to re-config cv-bridge or openCV to python 2.7, according to this link: https://stackoverflow.com/questions/49221565/unable-to-use-cv-bridge-with-ros-kinetic-and-python3?rq=1