PX4/PX4-SITL_gazebo-classic #232: cannot find mavlink_sitl_gazebo

FaboNo, 2018-09-19:
I installed px firmware under ws/src/Firmware and `make posix_sitl_default gazebo`
in the catkin_ws I have also mavros installed without any issue
I ran catkin build on the root as well.

I can see that sitl_gazebo has been build but unfortunately I cannot find mavlink_sitl_gazebo (not in catkin_ws/build or usr:share/gazebo-7)

so when I launch:
roslaunch px4 mavros_posix_sitl.launch
Gazebo is starting with an empty world and a Iris model but I do not know from where there coming (if I modify catkin_ws/src/Firmware/Tools/sitl_gazebo/worlds/iris.world  i do see anychange)
Can someone can tell me what am I missing?
Thank you

EliaTarasov, 2018-09-21:
Could you specify more accurate what problem do you have?
Seems like you did everything ok (except maybe executing `source Tools/setup_gazebo.bash $(pwd) $(pwd)/build/posix_sitl_default`).
`mavlink_sitl_gazebo` and `px4` are packages, not executables. 
Generally you don't `catkin_make` or `catkin build` under `Firmware` directory.
ROS already has `gazebo_ros` executable node which is called everytime you run `mavlink_sitl_gazebo` or `px4` packages.

FaboNo, 2018-09-30:
Thank you Elia for your answer
Fabrice