PX4/PX4-SITL_gazebo-classic #233: sitl_gazebo gimbal controller plugin performance/functionality

jannsta1, 2018-09-27:
I'm trying to use the ```gazebo_gimbal_controller_plugin.cpp``` to control a gimbal (as it says on the tin).  

I would like to know what the expected (default) behaviour of this gimbal controller is? 

Reading through the code I assume that MAV roll and pitch motions should be cancelled (the gimbal maintains a setpoint relative to the world) and gimbal yaw is held at the previous value?

The above behaviour would meet my requirements well but its not really what I see in the simulator. I've tried operating the typhoon_h480 model which is already has an attached gimbal model and the gimbal seems pretty wayward (e.g. if I run ```./test/rostest_px4_run.sh mavros_posix_tests_offboard_posctl.test gui:=true``` from the Firmware folder the camera view doesn't stay level, hard to tell what cancellation is happening).

In summary: I'm wondering if I need to work on my setup more or I have the wrong idea about what the gimbal controller plugin functionality is.

jannsta1, 2018-09-28:
Seems to be working much better if I hardcode the joint setpoints (this->rollCommand etc) in the plugin. I guess I just need to work on getting the relevant topics to the plugin.

closed 2018-09-28 by jannsta1

vinay-lanka, 2023-02-27:
Hey!
Did you have any luck modifying the plugin to improve the gimbal performance? I did try to change PID values but I wasn't able to improve much.