PX4/PX4-SITL_gazebo-classic #241: Question about GazeboMotorModel::UpdateForcesAndMoments

yota-code, 2018-10-28:
I'm sorry to ask this here, but I hope you would be able to redirect me to some relevant sources to find an answer...

I'm playing with a toy-controller-project with the help of the sitl_gazebo simulation, and I don't understand why in the function `void GazeboMotorModel::UpdateForcesAndMoments()` the parameter `motor_rot_vel_` is approximately 1.1519 times smaller than the velocity set somewhere else via the `cmd.set_motor_speed(<motor #>, <commanded speed in RPM>)` and which correspond exactly to `ref_motor_rot_vel_`

So, what is `motor_rot_vel_` ? and how is it computed ? and why 1.1519 (?_?)

EliaTarasov, 2018-10-29:
@yota-code 

> the parameter real_motor_velocity is approximately 1.1519 times smaller than the velocity set somewhere else via the cmd.set_motor_speed(<motor #>, <commanded speed in RPM>)

Where do you see this: `cmd.set_motor_speed(<motor #>, <commanded speed in RPM>)`

You may want this setting to change: https://github.com/PX4/sitl_gazebo/blob/master/models/typhoon_h480/typhoon_h480.sdf#L1318

yota-code, 2018-10-29:
Sorry, the issue is so silly it's hard to explain (^o^;)

`cmd.set_motor_speed()` is the function I call at the end of my controller plugin, where `cmd` is defined as a `mav_msgs::msgs::CommandMotorSpeed` message

If I understand the role of the `rotorVelocitySlowdownSim` constant it has more to do with the visual representation...

The problem happens on `sitl_gazebo/gazebo_motor_plugin.cpp` file, line `194` where the `force` is computed with the value of `motor_rot_vel_` which is 15% slower than what I requested, and I don't understand why...

`double force = real_motor_velocity * real_motor_velocity * motor_constant_;`

Which means, I don't understand either why `motor_rot_vel_ = joint_->GetVelocity(0);` on line `189` return an erroneous value, while `ref_motor_rot_vel_ = std::min(static_cast<double>(rot_velocities->motor_speed(motor_number_)), static_cast<double>(max_rot_velocity_));` on line `181` has the right value.

EliaTarasov, 2018-10-29:
Here is filter that might affect velocity and produce this difference: https://github.com/PX4/sitl_gazebo/blob/master/src/gazebo_motor_model.cpp#L244

yota-code, 2018-10-29:
Well... after some investigations, here are the curves (cf. the attached picture)
from the line `244` where code is :

`ref_motor_rot_vel = rotor_velocity_filter_->updateFilter(ref_motor_rot_vel_, sampling_time_);`

`ref_motor_rot_vel_filtered` represents the values after the filter (or `ref_motor_rot_vel`) where
`ref_motor_rot_vel_raw` is the plot before the filter (or `ref_motor_rot_vel_`)

The filtering is done but the effect is marginal, and the mean value of the filter is zero. 

The green curve below represents `motor_rot_vel_` and `real_motor_velocity` respectively, as found in `GazeboMotorModel::UpdateForcesAndMoments()`

For a given time, we measure : 
- `ref_motor_rot_vel = 4965.9 RPM` which is `520.03 rad.s-1` where
- `motor_rot_vel_ = real_motor_velocity = 451.4 rad.s-1` ... which is 1.1520 times lower (almost 1.1519 :D) 

[image]

yota-code, 2018-10-31:
One last plot to be clear... And I'll let this issue open until I find the source of the problem :)

In blue, `real_motor_velocity` as used line`194` to compute the force.
In orange, the `ref_motor_velocity` before the filter (light orange) and after (dark orange) for different speed values commanded across the full range...

And in between a 1.152... ratio

[image]

I'll be back ! Hopefully with an answer !

EliaTarasov, 2018-11-14:
@yota-code just a guess: maybe you have badly tuned pid-regulator on your engines joints? It may looks like P and I are too small.

EliaTarasov, 2018-11-15:
[image]
I tested this with standard `iris` model and it seems that difference between reference (commanded) velocity and real velocity is small enough

yota-code, 2018-11-15:
Thank you for investigating :)

It shouldn't be the PID, it is not used and ... the PID shouldn't be able to inject a constant shift.
I started with the Typhoon 480, modified to behave like a dji F550 and I must admit that I changed some of the parameters of the engines... but I still don't see what could explain that. (here is the sdf file [dji_f550.sdf.txt](https://github.com/PX4/sitl_gazebo/files/2584297/dji_f550.sdf.txt))
I still miss something, but I could cancel the difference by dividing beforehand by 0.8... to cancel the x0.8 :)

EliaTarasov, 2018-11-15:
Then i would advise you to revert back changes, and test this issue on the original `typhoon_h490`. Then modify it step by step and check for this issue on each step.

closed 2020-04-25 by LorenzMeier