PX4/PX4-SITL_gazebo-classic #242: Got stuck in "Performing build step for 'sitl_gazebo' "

Kumaron, 2018-10-31:
I just downloaded ubuntu_sim_ros_gazebo.sh and finished the installation. 
I tried to do px4 sitl simulation by `make posix_sitl_default gazebo`, it got stuck in `Performing build step for 'sitl_gazebo`. It took a very long time to get into gazebo. Is there any solution that can solve this? 
And when I tried it the second time, it just got stuck in `Performing build step for 'sitl_gazebo` and cannot get into gazebo.

Kumaron, 2018-10-31:
`kumaron@ubuntu:~/src/Firmware$ make posix_sitl_default gazebo
ninja: Entering directory `/home/kumaron/src/Firmware/build/posix_sitl_default'
[2/6] Performing configure step for 'sitl_gazebo'
-- install-prefix: /usr/local
-- Boost version: 1.58.0
-- Found the following Boost libraries:
--   system
--   thread
--   timer
--   chrono
--   date_time
--   atomic
-- Boost version: 1.58.0
-- Building klt_feature_tracker without catkin
-- Building OpticalFlow with OpenCV
-- catkin DISABLED
Gazebo version: 7.14
-- Using C++17 compiler
-- Configuring done
-- Generating done
-- Build files have been written to: /home/kumaron/src/Firmware/build/posix_sitl_default/build_gazebo
[3/6] Performing build step for 'sitl_gazebo'
`
Above is how it shows when I try to run `make posix_sitl_default gazebo`, and it just stops in `[3/6] Performing build step for 'sitl_gazebo'`.

Kumaron, 2018-10-31:
Emmmm it works at last, maybe because it was loading resources and my ubuntu PC is too old?

LorenzMeier, 2018-10-31:
I'm closing this for now. Also `make distclean` can help to clean up your installed state.

closed 2018-10-31 by LorenzMeier