PX4/PX4-SITL_gazebo-classic #247: tinyxml2 dependency problem

julianoes, 2018-11-19:
I'm getting the following build error on Arch Linux with latest master (2adc86b5bf84543091470966caf33c3b4f3ee666).

```
gz: error while loading shared libraries: libtinyxml2.so.6: cannot open shared object file: No such file or directory
make[7]: *** [CMakeFiles/sdf.dir/build.make:65: /home/julianoes/src/Firmware-master/Tools/sitl_gazebo/models/iris/iris.sdf] Error 127
```

TSC21, 2018-11-19:
@julianoes did you confirm that you have libtinyxml2 on your Arch dist?
We probably need to set a new parallel build for arch in CI to make sure we have building on the most common OS's.

julianoes, 2018-11-19:
I have `tinyxml2`, and I can see that the linking error is somehow from `gz`, so that's confusing.

TSC21, 2018-11-19:
So, out of our hands? Maybe something to report on the Gazebo bitbucket repo?

julianoes, 2018-11-19:
Ok I get it. Gazebo tries to access the wrong lib version.
```
ls -l /usr/lib | grep tinyxml2
lrwxrwxrwx 1 root root        16 Nov 18 17:46 libtinyxml2.so -> libtinyxml2.so.7
lrwxrwxrwx 1 root root        20 Nov 18 17:46 libtinyxml2.so.7 -> libtinyxml2.so.7.0.1
-rwxr-xr-x 1 root root    100312 Nov 18 17:46 libtinyxml2.so.7.0.1
```

Thanks @TSC21.

closed 2018-11-19 by julianoes

TSC21, 2018-11-19:
Np. Glad you figured it out!

julianoes, 2018-11-19:
To follow up with this: even though I recompiled gazebo after the tinyxml2 update, it still had reference to the old library. Actually needed was a recompilation of `ignition-common`.