PX4/PX4-SITL_gazebo-classic #254: Error while using matrice 100 model

samarth8927, 2018-12-03:
While adding model to gazebo for Matrice -100 I got the following error:
Error [parser.cc:286] Error finding file [/home/sam_ros/src/sitl_gazebo/models/matrice_100/matrice_100-gen.sdf].

TSC21, 2018-12-03:
@samarth8927 what procedures did you follow to "add" the model? The Matrice 100 model is generated on build time using a jinja generator Python script. Did you build the repository?

samarth8927, 2018-12-03:
I am utilizing the insert tab of gazebo. I think the issue is with .sdf.jinja files as I am able to insert .sdf based models. I am actually new to gazebo, I did build sitl_gazebo repository. Can you please tell how to use a jinja generator python script, as i am unable to find much information with this regard on internet.