PX4/PX4-SITL_gazebo-classic #266: Increase camera FPS without breaking physics simulation

ChrisHughes, 2019-01-11:
Gazebo 9, PX4 SITL 1.8.0

I'm running into an issue when using SITL simulation with ROS and adding cameras to my model. I'm running a vision software which benefits from a higher frame rate. I do not need the simulation to run in real time.

In my physics definition it appears that max_step_size and real_time_update_rate control the number of cycles in the loop dedicated to sending data to PX4 and handling the camera. Whenever I make a change that benefits camera framerate, it seems highly detrimental to the simulation, with either in-air stability becoming an issue, or PX4 complaining about stale sensor data.

Are there any ways around this? Such as separate physics model for PX4 and cameras? Or decoupling the cameras or PX4? Or any other workaround to get it such that the framerate per simulated second of flight (not real world fps) is improved.

ChrisHughes, 2019-02-24:
I managed to solve this by turning off shadow rendering. This can be done in the world file by placing the following:

```
    <scene>
      <shadows>false</shadows>
    </scene>
```

LorenzMeier, 2019-02-24:
Cool - we should default to that then as the shadow has no real merit for the simulation.

LorenzMeier, 2019-02-24:
@mrivi Could you help adding that to all models by default?

TSC21, 2019-02-24:
If we are to get better visual features on a camera pipeline, does it make sense to actually turn off the shadows on the scenes? I think that setting those off by default it's not the ideal solution.

ChrisHughes, 2019-02-24:
At least add it to the documentation in an obvious way, as this was not an obvious fix. I'm running a new core i7 with a GTX 1050 and literally every other 3D visualizser, from RViz to RTabMap runs great. So what's up with Gazebo running at 1fps if I'm lucky then running at about 5-10fps if I turn off shadows? Something very inefficient with the render pipeline there.

mrivi, 2019-02-25:
@ChrisHughes fix in #288

TSC21, 2020-03-10:
A possible solution was presented at #288. But as explained on the PR, doesn't make sense to default this, unless really required.

closed 2020-03-10 by TSC21