PX4/PX4-SITL_gazebo-classic #275: Exaggerated vibration when hovering with Solo gazebo simulation

diegomrt, 2019-02-04:
I've noticed that there is a huge vibration in "roll" when launching the default 3DR Solo Gazebo simulation with `make posix gazebo_solo` I've tested it in two different computers, getting the same issue. However, the 3DR Iris simulation doesn't have such exaggerated vibration. 

[I've uploaded a video here](https://www.youtube.com/watch?v=DojFHlTVwn8)

Is it a code "bug" or a PX4 parameter (noise-related) that is incorrectly set? Default values are used in all 3DR Solo parameters and in PX4.  

Thanks in advance.

**$make posix gazebo_solo**
ninja: Entering directory `/home/diego/Firmware/build/px4_sitl_default'
[2/6] Performing configure step for 'sitl_gazebo'
-- install-prefix: /usr/local
-- Boost version: 1.58.0
-- Found the following Boost libraries:
--   system
--   thread
--   timer
--   chrono
--   date_time
--   atomic
-- Boost version: 1.58.0
-- Building klt_feature_tracker without catkin
-- Building OpticalFlow with OpenCV
-- catkin DISABLED
Gazebo version: 7.0
-- Using C++14 compiler
-- Configuring done
-- Generating done
-- Build files have been written to: /home/diego/Firmware/build/px4_sitl_default/build_gazebo
[3/6] Performing build step for 'sitl_gazebo'
ninja: no work to do.
[6/6] cd /home/diego/Firmware/build/px.../diego/Firmware/build/px4_sitl_default
SITL ARGS
sitl_bin: /home/diego/Firmware/build/px4_sitl_default/bin/px4
debugger: none
program: gazebo
model: solo
src_path: /home/diego/Firmware
build_path: /home/diego/Firmware/build/px4_sitl_default
GAZEBO_PLUGIN_PATH :/home/diego/Firmware/build/px4_sitl_default/build_gazebo
GAZEBO_MODEL_PATH :/home/diego/Firmware/Tools/sitl_gazebo/models
LD_LIBRARY_PATH /home/diego/catkin_ws/devel/lib:/opt/ros/kinetic/lib:/opt/ros/kinetic/lib/x86_64-linux-gnu:/home/diego/Firmware/build/px4_sitl_default/build_gazebo
Gazebo multi-robot simulator, version 7.0.0
Copyright (C) 2012-2016 Open Source Robotics Foundation.
Released under the Apache 2 License.
http://gazebosim.org

[Msg] Waiting for master.
[Msg] Connected to gazebo master @ http://127.0.0.1:11345
[Msg] Publicized address: 192.168.1.5
[Dbg] [gazebo_mavlink_interface.cpp:137] <joint_name> not found for channel[0] no joint control will be performed for this channel.
[Dbg] [gazebo_mavlink_interface.cpp:137] <joint_name> not found for channel[1] no joint control will be performed for this channel.
[Dbg] [gazebo_mavlink_interface.cpp:137] <joint_name> not found for channel[2] no joint control will be performed for this channel.
[Dbg] [gazebo_mavlink_interface.cpp:137] <joint_name> not found for channel[3] no joint control will be performed for this channel.
[Msg] Conecting to PX4 SITL using TCP
[Msg] Lockstep is enabled
[Msg] Speed factor set to: 1
SITL COMMAND: "/home/diego/Firmware/build/px4_sitl_default/bin/px4" "/home/diego/Firmware"/ROMFS/px4fmu_common -s etc/init.d-posix/rcS -t "/home/diego/Firmware"/test_data
INFO  [px4] Creating symlink /home/diego/Firmware/ROMFS/px4fmu_common -> /home/diego/Firmware/build/px4_sitl_default/tmp/rootfs/etc
0 WARNING: setRealtimeSched failed (not run as root?)

______  __   __    ___ 
| ___ \ \ \ / /   /   |
| |_/ /  \ V /   / /| |
|  __/   /   \  / /_| |
| |     / /^\ \ \___  |
\_|     \/   \/     |_/

px4 starting.

INFO  [px4] Calling startup script: /bin/sh etc/init.d-posix/rcS 0
INFO  [dataman] Unknown restart, data manager file './dataman' size is 11405132 bytes
INFO  [simulator] Waiting for simulator to connect on TCP port 4560
INFO  [simulator] Simulator connected on TCP port 4560.
[Msg] Using MAVLink protocol v2.0
Gazebo multi-robot simulator, version 7.0.0
Copyright (C) 2012-2016 Open Source Robotics Foundation.
Released under the Apache 2 License.
http://gazebosim.org

[Msg] Waiting for master.
[Msg] Connected to gazebo master @ http://127.0.0.1:11345
[Msg] Publicized address: 192.168.1.5
INFO  [init] Mixer: etc/mixers/quad_x.main.mix on /dev/pwm_output0
INFO  [mavlink] mode: Normal, data rate: 4000000 B/s on udp port 14570 remote port 14550
INFO  [mavlink] mode: Onboard, data rate: 4000000 B/s on udp port 14580 remote port 14540
INFO  [logger] logger started (mode=all)
INFO  [mavlink] MAVLink only on localhost (set param MAV_BROADCAST = 1 to enable network)
INFO  [logger] Start file log (type: full)
INFO  [logger] Opened full log file: ./log/2019-02-03/21_57_38.ulg
INFO  [px4] Startup script returned successfully
pxh> INFO  [mavlink] partner IP: 127.0.0.1
INFO  [ecl/EKF] EKF aligned, (pressure height, IMU buf: 22, OBS buf: 14)
[Wrn] [Publisher.cc:140] Queue limit reached for topic /gazebo/default/solo/motors, deleting message. This warning is printed only once.
[Wrn] [Publisher.cc:140] Queue limit reached for topic /gazebo/default/solo/motor_speed/0, deleting message. This warning is printed only once.
[Wrn] [Publisher.cc:140] Queue limit reached for topic /gazebo/default/solo/motor_speed/1, deleting message. This warning is printed only once.
[Wrn] [Publisher.cc:140] Queue limit reached for topic /gazebo/default/solo/motor_speed/2, deleting message. This warning is printed only once.
[Wrn] [Publisher.cc:140] Queue limit reached for topic /gazebo/default/solo/motor_speed/3, deleting message. This warning is printed only once.
INFO  [ecl/EKF] EKF GPS checks passed (WGS-84 origin set)
INFO  [ecl/EKF] EKF commencing GPS fusion
INFO  [commander] Takeoff detected
INFO  [commander] Landing detected
INFO  [logger] closed logfile, bytes written: 6395899

TSC21, 2019-10-11:
@diegomrt is this still an issue to you?

diegomrt, 2019-10-11:
No @TSC21, it can be closed.

closed 2019-10-11 by diegomrt