PX4/PX4-SITL_gazebo-classic #278: Question about magnetic field data

EliaTarasov, 2019-02-06:
[Magnetic filed data](https://github.com/PX4/sitl_gazebo/blob/master/src/gazebo_mavlink_interface.cpp#L283) provided for Zurich states that D-component has negative sign. However [WMM2015 Online Calculator](https://www.ngdc.noaa.gov/geomag/calculators/magcalc.shtml#igrfwmm) provides positive sign. Reverse of this sign seem has no affect on flight performance but anyway it's good to fix it if it's wrong. What's the reason for this?

julianoes, 2019-02-15:
Do you know the coordinate frames of the source and the field needed in sitl_gazebo?

EliaTarasov, 2019-02-15:
According [to](https://github.com/PX4/sitl_gazebo/blob/master/src/gazebo_mavlink_interface.cpp#L599) and [to](https://github.com/PX4/sitl_gazebo/blob/master/src/gazebo_mavlink_interface.cpp#L624) the only rotations applied are declination and successive rotation to body. So seems like the source `mag_d_` frame must be the NED.

CarlOlsson, 2019-02-15:
Why not get the strength and inclination also from the geo lookup tables?

EliaTarasov, 2019-02-15:
@CarlOlsson That's much better solution. I'll add this then.

julianoes, 2019-02-16:
I'd say copy it in.

EliaTarasov, 2019-02-18:
Done.

EliaTarasov, 2019-02-18:
Closed by [PR](https://github.com/PX4/sitl_gazebo/pull/282)

closed 2019-02-18 by EliaTarasov