PX4/PX4-SITL_gazebo-classic #280: (question/bug)-possible error on iris inertia matrix

BazookaJoe1900, 2019-02-08:
Hi,
I have played with the iris model and tried to find how the inertia matrix of the main body has been calculated on it. I am not sure about it, but I think that the matix is 'impossible'.
I have tried to find 'equivalent' box for drone inertia, and got that d length of the box is immposible to calculate:

I	ixx=	0.0347563  =m/12*(w*w+d*d)
II	iyy=	0.0458929  =m/12*(h*h+d*d)
III	izz=	0.0977        =m/12*(h*h+w*w)
IIII    m = 1.5

I		0.2780504 =w*w+d*d
II		0.3671432 =h*h+d*d
III		0.7816	 =h*h+w*w

I+II-III		-0.1364064	=2*d*d
last line is immposible, negative d*d....

can anyone explain?

EliaTarasov, 2019-02-11:
Pretty much all of default models for `gazebo_sitl` have very inconsisted MoI. To check it, choose `View->Inertias' in gazebo visual client. Each inertia should visually match construction element (e.g. props, gears, fuselage).

BazookaJoe1900, 2019-02-12:
actually, for the iris, its looks that some of the MoI are well calculated on the xacro file.
for example:
https://github.com/PX4/sitl_gazebo/blob/f8b4f545f1b2d90000e37cc53f744849d939fb52/models/rotors_description/urdf/iris.xacro#L40

but the main body inertia is 'hard codded':
https://github.com/PX4/sitl_gazebo/blob/f8b4f545f1b2d90000e37cc53f744849d939fb52/models/rotors_description/urdf/iris.xacro#L34

As I wrote on the first comment, this inertia is immposible.

If you look a the gazebo GUI, as you suggested, there is no inertia of the body. the inertia you see is of the IMU.

closed 2019-05-18 by LorenzMeier