PX4/PX4-SITL_gazebo-classic #286: Failed to load plugin librotors_gazebo_odometry_plugin.so

ArkadiuszNiemiec, 2019-02-24:
I am testing the new `odometry` pipeline and cannot make it work even if I set `SEND_ODOMETRY_DATA` to `ON`. While debugging I have tried to set the `enable_ground_truth` to `true` and ended up with this message:

```[Err] [Plugin.hh:180] Failed to load plugin librotors_gazebo_odometry_plugin.so: librotors_gazebo_odometry_plugin.so: cannot open shared object file: No such file or directory```

I have no clue what's wrong but it's reproducible on all my machines.

TSC21, 2019-02-24:
You are making a confusion with the `rotors_simulator` package available plugin. There's no plugin as such on `sitl_gazebo`, and you can check that on both the `src` folder and the `CMakeLists.txt`.
The plugin we use to send visual odometry is called `gazebo_vision_plugin`.

closed 2019-02-24 by TSC21

ArkadiuszNiemiec, 2019-02-24:
@TSC21 So how do I enable the ground truth stream?

You also mentioned that there's an option to enable odometry stream, is that the `SEND_ODOMETRY_DATA` option? I have enabled it and checked that `mavlink_receiver` is not receiving any data. And I also get:
```pxh> commander takeoff
pxh> WARN  [mc_pos_control] Auto activation failed with error: Activation Failed
WARN  [mc_pos_control] Position-Ctrl activation failed with error: Activation Failed
WARN  [mc_pos_control] Altitude-Ctrl activation failed with error: Activation Failed
INFO  [commander] Takeoff detected
```

TSC21, 2019-02-24:
@ArkadiuszNiemiec it's not the  `mavlink_receiver` that receives the data coming from Gazebo when using SITL: it's `simulator_mavlink`.
Please do bring all questions that are not related to the topic to PX4 Slack or the PX4 forum.