PX4/PX4-SITL_gazebo-classic #287: Gazebo SITL and PX4 not handling simulated wind correctly for fixed wing

Antiheavy, 2019-02-24:
moving https://github.com/PX4/Firmware/issues/11319 to the appropriate repo.

based on a conversation with @TSC21 I suspect recent changes to the wind model are not properly accounting for fixed wing airspeed measurement and therefore wind estimation.

As pointed out by @CHschizo in this other issue post here: [#11116 (comment)](https://github.com/PX4/Firmware/issues/11116#issuecomment-452724619) the wind estimate and flight performance isn't working correctly in Gazebo SITL. See screenshot below of the vehicle tracking offset to the intended location east-west cross-track error, but tracking okay north-south cross-track error.

This screenshot is using the default PX4 Gazebo Plane model:
[image]

Antiheavy, 2019-02-24:
@TSC21 should we just disable the wind module for fixed wing?  or can this be fixed easily?

Jaeyoung-Lim, 2019-10-23:
@Antiheavy Are there any updates on this?

xdwgood, 2019-10-31:
@Antiheavy  I also encountered this problem, how can I set the simulation to wind = 0? thanks

Antiheavy, 2019-10-31:
@Jaeyoung-Lim @xdwgood I just made a PR that shows how I worked around this.  I don't know if this PR will be accepted, but it worked for me.
https://github.com/PX4/sitl_gazebo/pull/364

Antiheavy, 2019-11-26:
It looks like @Jaeyoung-Lim is working on a true fix to this problem in this PR: https://github.com/PX4/sitl_gazebo/pull/375

nikmad, 2020-02-28:
What is the difference between wind in 'windDirection' and wind gust in this plugin? Are these both parameters referring to 'wind gust' that comes erratically on top of normal relative wind as seen from the aircraft by virtue of its relative velocity?
      <windDirection>0 1 0</windDirection>
      <windGustDirection>0 0 0</windGustDirection>

nikmad, 2020-02-28:
What is the difference between **wind** in 'windDirection' and **wind gust** in this plugin? Are these both parameters referring to 'wind gust' that comes erratically on top of normal relative wind as seen from the aircraft by virtue of its relative velocity?
>  <windDirection>0 1 0</windDirection>
> <windGustDirection>0 0 0</windGustDirection>

nikmad, 2020-02-28:
What is the difference between wind in 'windDirection' and wind gust in 'windGustDirection' in this plugin? Are these both parameters referring to 'wind gust' that comes erratically on top of normal relative wind as seen from the aircraft by virtue of its relative velocity?

LorenzMeier, 2020-04-25:
@Jaeyoung-Lim Got this fixed in the interim?

Jaeyoung-Lim, 2020-04-25:
@LorenzMeier Yes, this should have been fixed with https://github.com/PX4/sitl_gazebo/pull/375

closed 2020-04-25 by Jaeyoung-Lim