PX4/PX4-SITL_gazebo-classic #289: Abort on startup because port is occupied

julianoes, 2019-02-25:
Since the introduction of lockstep (#257) we see an abort on startup quite often:

```
(1551081994 550255675) [Err] [gazebo_mavlink_interface.cpp:416] bind failed: Address already in use, aborting
```

The problem is that we can't bind to the port if it is already in use. However, it should not happen in the first place that the port is already in use (unless Gazebo SITL is already running).
My guess is that we either don't close the port properly, or we don't shutdown gzserver correctly and therefore the port does not get cleaned up properly.

Initially reported by @mrivi.

shrit, 2019-03-24:
I am having this problem most of the time, in most cases it is due to the fact that gazebo server is still running. I need to kill it manually to be able close the used ports

closed 2019-05-09 by TSC21

julianoes, 2019-05-09:
> I need to kill it manually to be able close the used ports

If it's still running, there is not much we can do. The PX4 SITL scripts usually clean up when you do Ctrl+C or type `shutdown`. And in this case it should all get cleaned up now.

PR #316 (2019-05-09): Add MAVLink forwarding for SDK/MAVROS for Gazebo HITL
This enables MAVLink forwarding on UDP port 14540 for Gazebo HITL.

~The current state is working but has quite a bit of duplication, so I'm planning to refactor this a bit and clean it up.~
I think for the sake of not having too much churn I suggest to merge this as is, and maybe refactor it later.

With this PR Gazebo HITL actually works as described in the docs already :smile:, FYI @hamishwillee.

Also, this disables the `TCP TIME_WAIT` state on gzserver shutdown and resolves #289.