PX4/PX4-SITL_gazebo-classic #290: World rubble not working

julianoes, 2019-02-25:
According to @mrivi the world rubble is not working:

> I noticed that the world `rubble` is not working. @TSC21 do you know where it's used?

@mrivi how can you reproduce "not working", how do you choose rubble?

mrivi, 2019-02-25:
I launch it from ros where `world_path` is the path to `rubble.world`

```bash
<include file="$(find gazebo_ros)/launch/empty_world.launch">
        <arg name="world_name" value="$(arg world_path)" />
    </include>
```

First I noticed that the `real_time_update_rate` was not set properly to work with the lockstep (same issues we had with the avoidance. it's 1000 instead of 250). But even fixing that, the world never loads into gazebo.

julianoes, 2019-02-25:
What about the other params, not just `real_time_update_rate` but also the `max_step_size`?

And where is this file?

mrivi, 2019-02-25:
I was wrong. I can load it 
[image]

but 
```bash
WARN  [ecl/EKF] EKF baro hgt timeout - reset to baro
WARN  [ecl/EKF] EKF baro hgt timeout - reset to baro
WARN  [ecl/EKF] EKF baro hgt timeout - reset to baro
WARN  [ecl/EKF] EKF GPS fusion timeout - reset to GPS
WARN  [ecl/EKF] EKF baro hgt timeout - reset to baro
WARN  [ecl/EKF] EKF baro hgt timeout - reset to baro
WARN  [ecl/EKF] EKF baro hgt timeout - reset to baro
WARN  [ecl/EKF] EKF baro hgt timeout - reset to baro
WARN  [ecl/EKF] EKF baro hgt timeout - reset to baro
WARN  [ecl/EKF] EKF baro hgt timeout - reset to baro
WARN  [ecl/EKF] EKF GPS fusion timeout - reset to GPS
WARN  [ecl/EKF] EKF GPS fusion timeout - reset to GPS
WARN  [ecl/EKF] EKF baro hgt timeout - reset to baro
WARN  [ecl/EKF] EKF baro hgt timeout - reset to baro
WARN  [ecl/EKF] EKF baro hgt timeout - reset to baro
WARN  [ecl/EKF] EKF baro hgt timeout - reset to baro
WARN  [ecl/EKF] EKF baro hgt timeout - reset to baro
WARN  [ecl/EKF] EKF baro hgt timeout - reset to baro
WARN  [ecl/EKF] EKF baro hgt timeout - reset to baro
WARN  [ecl/EKF] EKF GPS fusion timeout - reset to GPS
WARN  [commander] Arming DENIED. Manual throttle non-zero.
WARN  [ecl/EKF] EKF GPS fusion timeout - reset to GPS
WARN  [commander] Preflight Fail: Position unknown
WARN  [commander] Preflight Fail: Accel Range, hold still on arming
WARN  [commander] Preflight Fail: Accel Range, hold still on arming
```

You can launch it `roslaunch px4 posix_sitl.launch world:=$(find mavlink_sitl_gazebo)/worlds/rubble.world`

TSC21, 2019-10-11:
@mrivi is this still an issue?

RicardoM17, 2020-04-08:
I assume that now this is no longer an issue considering it was removed in #453

@julianoes this can probably be closed.

closed 2020-04-09 by julianoes