PX4/PX4-SITL_gazebo-classic #291: has no member named 'velocity_covariance'

jungmineeee, 2019-02-25:
I followed the commands. also, append the 'exports ' path in the ~/.bashrc file

[image]
(I'm using the newest VirtualBox and Ubuntu 16.04 LTS version, I set the VirtualBox for sharing the clipboard but it doesn't work. So I put these pictures)

When I do the command "make", there's an error raised. 

[image]

Kchour, 2019-02-25:
I have the exact error as well

mhkabir, 2019-02-25:
@TSC21

Kchour, 2019-02-25:
So I solved my issue by including the following repo files (https://github.com/mavlink/c_library_v2) into **/usr/local/mavlink/v2.0**

Make sure you extract all the files like so...

[image]

TSC21, 2019-02-26:
This only occurs when `sitl_gazebo` CMake fetches the Mavlink headers from the `ros-..-mavlink` install. When it's not updated, results on the above error. So or you take @Kchour solution or you need to wait for the `mavlink-gbp-release` to come out on the ROS build farm and then released as a deb package.

closed 2019-02-26 by TSC21