PX4/PX4-SITL_gazebo-classic #292: libgazebo_opticalflow_plugin.so segfault

Seanmatthews, 2019-02-26:
I've had problems with `libgazebo_opticalflow_plugin.so` while using gazebo9, PX4 master, Ubuntu 16.04.5. Trying to spawn the iris_opt_flow or iris_rplidar gets me a segfault with backtrace:

> #0 0x0000000000000000 in ?? ()
> #1 0x00007fffd1cc89b8 in cv::Mat::copyTo(cv::_OutputArray const&) const () from /usr/lib/x86_64-linux-gnu/libopencv_core.so.2.4 #2 0x00007ffef1df5c75 in trackFeatures(cv::Mat const&, cv::Mat const&, std::vector<cv::Point_, std::allocator<cv::Point_ > >&, std::vector<cv::Point_, std::allocator<cv::Point_ > >&, std::vector<int, std::allocator >&, int) () from /home/dev/src/Firmware/build/px4_sitl_default/build_gazebo/OpticalFlow/libOpticalFlow.so
> #3 0x00007ffef1df30f5 in OpticalFlowOpenCV::calcFlow(unsigned char*, unsigned int const&, int&, float&, float&) () from /home/dev/src/Firmware/build/px4_sitl_default/build_gazebo/OpticalFlow/libOpticalFlow.so
> #4 0x00007fff049bcc86 in gazebo::OpticalFlowPlugin::OnNewFrame(unsigned char const*, unsigned int, unsigned int, unsigned int, std::__cxx11::basic_string<char, std::char_traits, std::allocator > const&) () from /home/dev/src/Firmware/build/px4_sitl_default/build_gazebo/libgazebo_opticalflow_plugin.so
> #5 0x00007fff049cf900 in boost::_mfi::mf5<void, gazebo::OpticalFlowPlugin, unsigned char const*, unsigned int, unsigned int, unsigned int, std::__cxx11::basic_string<char, std::char_traits, std::allocator > const&>::operator()(gazebo::OpticalFlowPlugin*, unsigned char const*, unsigned int, unsigned int, unsigned int, std::__cxx11::basic_string<char, std::char_traits, std::allocator > const&) const () from /home/dev/src/Firmware/build/px4_sitl_default/build_gazebo/libgazebo_opticalflow_plugin.so
> #6 0x00007fff049ce5db in void boost::_bi::list6<boost::_bi::valuegazebo::OpticalFlowPlugin*, boost::arg<1>, boost::_bi::value, boost::_bi::value, boost::_bi::value, boost::_bi::value<std::__cxx11::basic_string<char, std::char_traits, std::allocator > > >::operator()<boost::_mfi::mf5<void, gazebo::OpticalFlowPlugin, unsigned char const*, unsigned int, unsigned int, unsigned int, std::__cxx11::basic_string<char, std::char_traits, std::allocator > const&>, boost::_bi::list5<unsigned char const* const&, unsigned int const&, unsigned int const&, unsigned int const&, std::__cxx11::basic_string<char, std::char_traits, std::allocator > const&> >(boost::_bi::type, boost::_mfi::mf5<void, gazebo::OpticalFlowPlugin, unsigned char const*, unsigned int, unsigned int, unsigned int, std::__cxx11::basic_string<char, std::char_traits, std::allocator > const&>&, boost::_bi::list5<unsigned char const* const&, unsigned int const&, unsigned int const&, unsigned int const&, std::__cxx11::basic_string<char, std::char_traits, std::allocator > const&>&, int) () from /home/dev/src/Firmware/build/px4_sitl_default/build_gazebo/libgazebo_opticalflow_plugin.so
> #7 0x00007fff049cbbe4 in void boost::_bi::bind_t<void, boost::_mfi::mf5<void, gazebo::OpticalFlowPlugin, unsigned char const*, unsigned int, unsigned int, unsigned int, std::__cxx11::basic_string<char, std::char_traits, std::allocator > const&>, boost::_bi::list6<boost::_bi::valuegazebo::OpticalFlowPlugin*, boost::arg<1>, boost::_bi::value, boost::_bi::value, boost::_bi::value, boost::_bi::value<std::__cxx11::basic_string<char, std::char_traits, std::allocator > > > >::operator()<unsigned char const*, unsigned int, unsigned int, unsigned int, std::__cxx11::basic_string<char, std::char_traits, std::allocator > const&>(unsigned char const*&&, unsigned int&&, unsigned int&&, unsigned int&&, std::__cxx11::basic_string<char, std::char_traits, std::allocator > const&) () from /home/dev/src/Firmware/build/px4_sitl_default/build_gazebo/libgazebo_opticalflow_plugin.so
> #8 0x00007fff049c811c in std::_Function_handler<void (unsigned char const*, unsigned int, unsigned int, unsigned int, std::__cxx11::basic_string<char, std::char_traits, std::allocator > const&), boost::_bi::bind_t<void, boost::_mfi::mf5<void, gazebo::OpticalFlowPlugin, unsigned char const*, unsigned int, unsigned int, unsigned int, std::__cxx11::basic_string<char, std::char_traits, std::allocator > const&>, boost::_bi::list6<boost::_bi::valuegazebo::OpticalFlowPlugin*, boost::arg<1>, boost::_bi::value, boost::_bi::value, boost::_bi::value, boost::_bi::value<std::__cxx11::basic_string<char, std::char_traits, std::allocator > > > > >::_M_invoke(std::_Any_data const&, unsigned char const*&&, unsigned int&&, unsigned int&&, unsigned int&&, std::__cxx11::basic_string<char, std::char_traits, std::allocator > const&) () from /home/dev/src/Firmware/build/px4_sitl_default/build_gazebo/libgazebo_opticalflow_plugin.so
> #9 0x00007fffedfd4117 in gazebo::rendering::Camera::PostRender() () from /usr/lib/x86_64-linux-gnu/libgazebo_rendering.so.9
> #10 0x00007ffff6038d50 in gazebo::sensors::CameraSensor::UpdateImpl(bool) () from /usr/lib/x86_64-linux-gnu/libgazebo_sensors.so.9
> #11 0x00007ffff6070d79 in gazebo::sensors::Sensor::Update(bool) () from /usr/lib/x86_64-linux-gnu/libgazebo_sensors.so.9
> #12 0x00007ffff60779a5 in gazebo::sensors::SensorManager::SensorContainer::Update(bool) () from /usr/lib/x86_64-linux-gnu/libgazebo_sensors.so.9
> #13 0x00007ffff6079ada in gazebo::sensors::SensorManager::Update(bool) () from /usr/lib/x86_64-linux-gnu/libgazebo_sensors.so.9
> #14 0x00007ffff7ba4157 in gazebo::Server::Run() () from /usr/lib/x86_64-linux-gnu/libgazebo.so.9
> #15 0x0000000000403032 in ?? ()
> #16 0x00007ffff62f4830 in __libc_start_main (main=0x402f30, argc=11, argv=0x7fffffffce68, init=, fini=, rtld_fini=, stack_end=0x7fffffffce58)
> at ../csu/libc-start.c:291
> #17 0x00000000004036a9 in _start ()

Also, I'll note that the reference to that library in iris_rplidar is spelled incorrectly. You'll need to correct the spelling to load the library and get the segfault.

julianoes, 2019-02-27:
Could you add the steps that you do to reproduce this? Thanks.

bozkurthan, 2019-03-25:
I have problem too with libgazebo_opticalflow_plugin.so. When I start `make posix gazebo_iris_rplidar` , I'm getting `Failed to load plugin libgazebo_opticalFlow_plugin.so: libgazebo_opticalFlow_plugin.so: cannot open shared object file: No such file or directory`

[image]

bozkurthan, 2019-03-25:
Re-download fixed the problem. Sorry for false alarm :+1:

julianoes, 2019-03-25:
@bozkurthan please create a new issue for the error you're seeing and include the command you're trying to run, information about your system, versions used, and copy paste output of the error. I think you're issue is not the same one.

bozkurthan, 2019-03-25:
@julianoes thanks. Sorry for misunderstanding. I'll create new issue according to your directions.

TSC21, 2019-10-11:
`iris_opt_flow` can be spawned and the optical flow plugins loads without any issue in Gazebo 9 currently. Please reopen if not the case. Closing.

closed 2019-10-11 by TSC21