PX4/PX4-SITL_gazebo-classic #296: Tiltrotor simulation when vehicle is disarmed

isidroas, 2019-03-06:
I've done the simulation of a tiltrotor and I've seen a strange behavorial. When the vehicle is disarmed, its rotors are waving. 
[image]

When I run 
pxh> listener actuator_outputs
I get this:
~~~~
   TOPIC: actuator_outputs instance 0 #1
    actuator_outputs_s
	   timestamp: 223652000  (0.000000 seconds ago)
	   noutputs: 11
	   output: [900.0000, 900.0000, 900.0000, 900.0000, 900.0000, 900.0000, 900.0000, 900.0000, 
   900.0000, 900.0000, 900.0000, nan, nan, nan, nan, nan]
~~~~

Is this this behavorial desired?

dagar, 2019-03-06:
No, that's not desired. I think this explains some of the intermittent failures we see in the PX4 CI system.

dlwalter, 2019-06-10:
I am seeing this too.  Servos are violently shaking on the Tiltrotor model in gazebo.  Is there any fix for this?

isidroas, 2019-06-14:
I achieved to solve that problem by changing the PID parameters of the motors. The first one is this [line](https://github.com/PX4/sitl_gazebo/blob/95adbb11f8deef06fb132bf0eb85c19e6ca00f6e/models/tiltrotor/tiltrotor.sdf#L1107)

Jaeyoung-Lim, 2020-01-06:
This has been fixed by https://github.com/PX4/sitl_gazebo/pull/386

@isidroas @dlwalter  Could you give it a try please?

closed 2020-01-16 by Jaeyoung-Lim