PX4/PX4-SITL_gazebo-classic #297: Passing distance sensor orientation through gazebo_mavlink_interface

Seanmatthews, 2019-03-08:
I'll refer to the issue I posted here: https://github.com/mavlink/mavros/issues/1203

Running the `typhoon_h480` model along with PX4 and MAVROS, it does not seem to pass a sonar sensor orientation through to MAVROS:

```
[/mavros][ERROR][extra_plugins::DistanceSensorPlugin::handle_distance_sensor][DS: sonar_1_pub: received sensor data has different orientation (NONE) than in config (PITCH_270)!]
[/mavros][ERROR][extra_plugins::DistanceSensorPlugin::handle_distance_sensor][DS: sonar_1_pub: received sensor data has different orientation (NONE) than in config (PITCH_270)!]
[/mavros][ERROR][extra_plugins::DistanceSensorPlugin::handle_distance_sensor][DS: sonar_1_pub: received sensor data has different orientation (NONE) than in config (PITCH_270)!]
[Wrn] [msgs.cc:1852] Conversion of sensor type[gpu_ray] not supported.
[Wrn] [msgs.cc:1852] Conversion of sensor type[sonar] not supported.
[Wrn] [msgs.cc:1852] Conversion of sensor type[gpu_ray] not supported.
[Wrn] [msgs.cc:1852] Conversion of sensor type[sonar] not supported.
[/mavros][ERROR][extra_plugins::DistanceSensorPlugin::handle_distance_sensor][DS: sonar_1_pub: received sensor data has different orientation (NONE) than in config (PITCH_270)!]
[/mavros][ERROR][extra_plugins::DistanceSensorPlugin::handle_distance_sensor][DS: sonar_1_pub: received sensor data has different orientation (NONE) than in config (PITCH_270)!]
[/mavros][ERROR][extra_plugins::DistanceSensorPlugin::handle_distance_sensor][DS: sonar_1_pub: received sensor data has different orientation (NONE) than in config (PITCH_270)!]
[/mavros][ERROR][extra_plugins::DistanceSensorPlugin::handle_distance_sensor][DS: sonar_1_pub: received sensor data has different orientation (NONE) than in config (PITCH_270)!]
```

Is there something that needs to be set in the gazebo_mavlink_interface section for the sonar that would pass the sensor properties along to MAVROS?

TSC21, 2019-03-09:
`gazebo_mavlink_interface` is sending `DISTANCE_SENSOR` messages with the orientation set to `MAV_SENSOR_ROTATION_PITCH_270`. I find it awkward that it's getting `NONE`. Are you properly set the sensor ID? Should be `0` on the `px4_config.yaml`.

Seanmatthews, 2019-03-13:
@TSC21 That is very awkward that `NONE` is meant as `PITCH_0`. 

The typhoon_h480 model throws this error by default, and it's the only included model that sports a sonar distance sensor by itself. But if the user is meant to change the `px4_config.yaml` settings in order to accommodate this and other vehicles regardless, we can safely close this issue. What do you think?

TSC21, 2019-10-11:
This was already handled in https://github.com/PX4/sitl_gazebo/pull/339 and https://github.com/PX4/sitl_gazebo/pull/340. Closing.

closed 2019-10-11 by TSC21

AlexWUrobot, 2022-12-20:
I checked the id is 0 but I still get the error like the below screenshot
DS: hrlv_ez4_pub: received sensor data has different orientation (NONE) than in config (PITCH_270)!
[image]
[image]