PX4/PX4-SITL_gazebo-classic #305: Cannot use iris_opt_flow for Auto Modes in Gazebo SITL

bozkurthan, 2019-03-25:
I've problem with optical flow camera in Gazebo Simulation. I tried some Flight modes in Simulation but most of them are not able to use. E.g. MISSION, HOLD. Only I can send TAKEOFF command and Optical Flow Camera works fine with this command. But somehow when I send MISSION or HOLD, Iris is rejecting these modes. Furthermore, I opened Analyze on QGroundControl and saw very different values between LOCAL_POSITION_NED.x and POSTION_TARGET_LOCAL_NED.x when I sent ALTITUDE mode command. Iris starts slowly freefall when I change mode from TAKEOFF to ALTITUDE

[image]
**POSTION_TARGET_LOCAL_NED.x  goes up because of mode change  from TAKEOFF to ALTITUDE**

Terminal LOGs:
`abnasuh@abnasuh-AS:~/src/Firmware$ sudo make posix gazebo_iris_opt_flow
[2/6] Performing configure step for 'sitl_gazebo'
-- install-prefix: /usr/local
-- Boost version: 1.58.0
-- Found the following Boost libraries:
--   system
--   thread
--   timer
--   filesystem
--   chrono
--   date_time
--   atomic
-- Boost version: 1.58.0
-- Found the following Boost libraries:
--   thread
--   system
--   filesystem
--   program_options
--   regex
--   iostreams
--   date_time
--   chrono
--   atomic
-- Boost version: 1.58.0
-- Looking for OGRE...
-- Found Ogre Ghadamon (1.9.0)
-- Looking for OGRE_Paging...
-- Found OGRE_Paging: optimized;/usr/lib/x86_64-linux-gnu/libOgrePaging.so;debug;/usr/lib/x86_64-linux-gnu/libOgrePaging.so
-- Looking for OGRE_Terrain...
-- Found OGRE_Terrain: optimized;/usr/lib/x86_64-linux-gnu/libOgreTerrain.so;debug;/usr/lib/x86_64-linux-gnu/libOgreTerrain.so
-- Looking for OGRE_Property...
-- Found OGRE_Property: optimized;/usr/lib/x86_64-linux-gnu/libOgreProperty.so;debug;/usr/lib/x86_64-linux-gnu/libOgreProperty.so
-- Looking for OGRE_RTShaderSystem...
-- Found OGRE_RTShaderSystem: optimized;/usr/lib/x86_64-linux-gnu/libOgreRTShaderSystem.so;debug;/usr/lib/x86_64-linux-gnu/libOgreRTShaderSystem.so
-- Looking for OGRE_Volume...
-- Found OGRE_Volume: optimized;/usr/lib/x86_64-linux-gnu/libOgreVolume.so;debug;/usr/lib/x86_64-linux-gnu/libOgreVolume.so
-- Looking for OGRE_Overlay...
-- Found OGRE_Overlay: optimized;/usr/lib/x86_64-linux-gnu/libOgreOverlay.so;debug;/usr/lib/x86_64-linux-gnu/libOgreOverlay.so
-- Found Protobuf: /usr/lib/x86_64-linux-gnu/libprotobuf.so (Required is at least version "2.3.0") 
-- Config-file not installed for ZeroMQ -- checking for pkg-config
-- Checking for module 'libzmq >= 4'
--   Found libzmq , version 4.1.4
-- Checking for module 'uuid'
--   Found uuid, version 2.27.0
-- Checking for module 'tinyxml2'
--   Found tinyxml2, version 2.2.0
-- Looking for dlfcn.h - found
-- Looking for libdl - found
-- FreeImage.pc not found, we will search for FreeImage_INCLUDE_DIRS and FreeImage_LIBRARIES
-- Checking for module 'gts'
--   Found gts, version 0.7.6
-- Checking for module 'libswscale'
--   Found libswscale, version 3.1.101
-- Checking for module 'libavdevice >= 56.4.100'
--   Found libavdevice , version 56.4.100
-- Checking for module 'libavformat'
--   Found libavformat, version 56.40.101
-- Checking for module 'libavcodec'
--   Found libavcodec, version 56.60.100
-- Checking for module 'libavutil'
--   Found libavutil, version 54.31.100
-- Checking for module 'jsoncpp'
--   Found jsoncpp, version 1.7.2
-- Checking for module 'yaml-0.1'
[... 93 lines cut ...]
Number of good matches: 3, desired: 3
Number of good matches: 3, desired: 3
Number of good matches: 2, desired: 2
Number of good matches: 3, desired: 3
Number of good matches: 2, desired: 2
Number of good matches: 4, desired: 4
Number of good matches: 3, desired: 3
Number of good matches: 5, desired: 5
Number of good matches: 3, desired: 3
Number of good matches: 1, desired: 9
Number of good matches: 1, desired: 16
Number of good matches: 1, desired: 15
INFO  [commander] Landing detected
INFO  [logger] closed logfile, bytes written: 3314637
[Wrn] [Publisher.cc:141] Queue limit reached for topic /gazebo/default/iris_opt_flow/motor_speed/1, deleting message. This warning is printed only once.
[Wrn] [Publisher.cc:141] Queue limit reached for topic /gazebo/default/iris_opt_flow/motors, deleting message. This warning is printed only once.
[Wrn] [Publisher.cc:141] Queue limit reached for topic /gazebo/default/iris_opt_flow/motor_speed/2, deleting message. This warning is printed only once.
[Wrn] [Publisher.cc:141] Queue limit reached for topic /gazebo/default/iris_opt_flow/motor_speed/3, deleting message. This warning is printed only once.
WARN  [commander] MANUAL CONTROL LOST (at t=196748ms)
WARN  [commander] MANUAL CONTROL LOST (at t=205880ms)
WARN  [commander] MANUAL CONTROL LOST (at t=235700ms)
`

Flight log with same state but another flight:
https://review.px4.io/plot_app?log=e80b5741-550d-4d93-b04b-9395d53e5197

bozkurthan, 2019-03-26:
OK. The problem is that mission needs Home Position for flight. On the other hand, I don't know why Iris doesn't receive HOLD mode without GPS.

In order to perform Mission I changed parameter that named EKF2_AID_MASK. 
Before change:
[image]

After change:
[image]

And waited for home position. Then, closed gps from terminal.
[image]

After all this step, Iris accepted HOLD and MISSION modes.

The whole steps should be like this or is this problem?

lamping7, 2019-07-10:
I think it's a problem. Mentioned here --> https://github.com/PX4/Firmware/pull/11665

bozkurthan, 2019-10-16:
I'm closing thanks @lamping7

closed 2019-10-16 by bozkurthan