PX4/PX4-SITL_gazebo-classic #312: Mistakes in formulas in gazebo_motor_model

srmx, 2019-04-09:
In file `gazebo_motor_model.cpp` I found several mistakes. 

1. In [line 163](https://github.com/LorenzMeier/sitl_gazebo/blob/afeae4c68725fafe367128bb1e6e9b644f74d766/src/gazebo_motor_model.cpp#L163) there must be a scalar product, not element-wise
2. In [line 153](https://github.com/LorenzMeier/sitl_gazebo/blob/afeae4c68725fafe367128bb1e6e9b644f74d766/src/gazebo_motor_model.cpp#L153) there is a coefficient called `motor_constant_`. Here must be `rotor_drag_coefficient_`, not `motor_constant_`.
3. In [line 164](https://github.com/LorenzMeier/sitl_gazebo/blob/afeae4c68725fafe367128bb1e6e9b644f74d766/src/gazebo_motor_model.cpp#L164) 
there must be a distinct coefficient, not `rotor_drag_coefficient_`.

It is pointed there that the formulas are taken from the paper "The True Role of Accelerometer Feedback in Quadrotor Control", but the formulas there are different.

TSC21, 2019-04-09:
@srmx can you support your review with actual tests? Cause the current math works for the current models, so if you can provide a test where you have this math modified and applied to the simulation, with correct results, that will help us support your evaluation. At this stage, just telling these are mistakes without tests is not going to help anyone.

srmx, 2019-04-09:
[The similar issue](https://github.com/ethz-asl/rotors_simulator/issues/256)  fixed 3 years ago in a different repo.

TSC21, 2019-04-09:
Makes sense. Can you provide a PR with a fix?

LorenzMeier, 2020-04-25:
@RomanBapst Could you please have a look?

Jaeyoung-Lim, 2020-10-05:
I believe these were addressed in https://github.com/PX4/sitl_gazebo/pull/484 

Please reopen if otherwise

closed 2020-10-05 by Jaeyoung-Lim