PX4/PX4-SITL_gazebo-classic #313: how to add the iris_down_cam model to the simulation environment

Oliver-ss, 2019-04-17:
I tried to add a camera model and combine it with the iris so that I could build a model named iris_down_cam. I followed the instruction and made some changes to the iris_base.xacro as you did. 
However, when I try to "make px4_sitl_default gazebo", error happened as: 
[Err] [Plugin.hh:165] Failed to load plugin libgazebo_ros_camera.so: libCameraPlugin.so: cannot open shared object file: No such file or directory
and if I directly "roslaunch px4 mavros_posix_sitl.launch", it would tell me 
Error: Unknown model 'iris_down_cam'
Error: Unknown model 'iris_fpv_cam'

Do you know what is the problem? Thank you very much!

vijayaganesh, 2019-04-21:
@Oliver-ss Did you find a solution to this issue? I'm facing the same problem too.

Oliver-ss, 2019-04-22:
> @Oliver-ss Did you find a solution to this issue? I'm facing the same problem too.

Yes, I directly change the iris.sdf. That is, I add a camera to the iris.sdf instead of make another iris model.

lamping7, 2019-07-15:
The problem is likely that you didn't create a startup file on the PX4 side to correspond with your model, but there isn't enough info provided to say anything more/useful.

closed 2020-04-25 by LorenzMeier